isaac-sim/IsaacLab #563: [Bug Report] Camera could not be initialized

ChengK321, 2024-06-25:
When I run the IsaacLab/source/standalone/tutorials/04_sensors/add_sensors_on_robot.py, there is an error: RuntimeError: Camera could not be initialized. Please ensure --enable_cameras is used to enable rendering.
The detial log is:
Starting kit application with the following args:  ['/home/stardust/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/stardust/simulation/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.kit', '--/app/tokens/exe-path=/home/stardust/simulation/IsaacLab/_isaac_sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/stardust/simulation/IsaacLab/_isaac_sim/exts', '--ext-folder', '/home/stardust/simulation/IsaacLab/_isaac_sim/apps', '--/renderer/activeGpu=0', '--/physics/cudaDevice=0', '--portable', '--/app/window/hideUi=False']
Passing the following args to the base kit application:  []
Loading user config located at: '/home/stardust/.local/share/ov/pkg/isaac-sim-4.0.0/kit/data/Kit/Isaac-Sim/4.0/user.config.json'
[Info] [carb] Logging to file: /home/stardust/.local/share/ov/pkg/isaac-sim-4.0.0/kit/logs/Kit/Isaac-Sim/4.0/kit_20240625_170653.log
2024-06-25 09:06:53 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.024s] [ext: omni.kit.async_engine-0.0.0] startup
[0.192s] [ext: omni.stats-0.0.0] startup
[0.193s] [ext: omni.client-1.1.0] startup
[0.208s] [ext: omni.datastore-0.0.0] startup
[0.208s] [ext: omni.blobkey-1.1.0] startup
[0.208s] [ext: omni.ujitso.default-1.0.0] startup
[0.208s] [ext: omni.hsscclient-0.0.0] startup
[0.209s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.223s] [ext: omni.assets.plugins-0.0.0] startup
[0.224s] [ext: omni.gpu_foundation-0.0.0] startup
[0.232s] [ext: carb.windowing.plugins-1.0.0] startup
2024-06-25 09:06:53 [227ms] [Warning] [carb.windowing-glfw.gamepad] Joystick with unknown remapping detected (will be ignored):  ASRock LED Controller [03000000ce260000a201000010010000]
[0.236s] [ext: omni.kit.renderer.init-0.0.0] startup
2024-06-25 09:06:53 [384ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Graphics (RPL-S)
2024-06-25 09:06:53 [384ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Graphics (RPL-S)

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.67        | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4070 SUPER    | Yes: 0 |     | 12282   MB | 10de      | 0          |
|     |                                  |        |     |            | 2783      | 047bceda.. |
|     |                                  |        |     |            | 1         |            |
|---------------------------------------------------------------------------------------------|
| 1   | Intel(R) Graphics (RPL-S)        |        |     | 23787   MB | 8086      | 0          |
|     |                                  |        |     |            | a780      | 868080a7.. |
|     |                                  |        |     |            | 0         |            |
|=============================================================================================|
| OS: 22.04.4 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.4, Kernel: 6.5.0-41-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: Intel(R) Core(TM) i7-14700K | Cores: 20 | Logical: 40
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31717 | Free Memory: 22390
| Total Page/Swap (MB): 31249 | Free Page/Swap: 31249
|---------------------------------------------------------------------------------------------|
[0.643s] [ext: omni.kit.pipapi-0.0.0] startup
[0.644s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.644s] [ext: omni.pip.compute-1.3.1] startup
[0.646s] [ext: omni.pip.cloud-1.1.3] startup
[0.648s] [ext: omni.isaac.core_archive-2.3.0] startup
[0.648s] [ext: omni.mtlx-0.1.0] startup
[0.648s] [ext: omni.usd.config-1.0.3] startup
[0.650s] [ext: omni.gpucompute.plugins-0.0.0] startup
[0.650s] [ext: omni.usd.libs-1.0.0] startup
[0.696s] [ext: omni.isaac.ml_archive-2.0.1] startup
[0.696s] [ext: omni.kit.loop-isaac-1.2.0] startup
[0.697s] [ext: omni.kit.test-0.0.0] startup
[0.745s] [ext: omni.kit.telemetry-0.5.0] startup
[0.757s] [ext: omni.appwindow-1.1.8] startup
[2.651s] [ext: omni.kit.renderer.core-1.0.1] startup
[16.757s] [ext: omni.kit.renderer.capture-0.0.0] startup
[16.759s] [ext: omni.kit.renderer.imgui-1.0.1] startup
[16.794s] [ext: omni.ui-2.23.11] startup
[16.799s] [ext: omni.kit.mainwindow-1.0.3] startup
[16.799s] [ext: carb.audio-0.1.0] startup
[16.801s] [ext: omni.uiaudio-1.0.0] startup
[16.801s] [ext: omni.kit.uiapp-0.0.0] startup
[16.801s] [ext: omni.usd.schema.semantics-0.0.0] startup
[16.863s] [ext: omni.usd.schema.isaac-2.0.2] startup
[16.866s] [ext: omni.usd.schema.scene.visualization-2.0.2] startup
[... 363 lines cut ...]
    main()
  File "/home/stardust/simulation/IsaacLab/source/standalone/tutorials/04_sensors/add_sensors_on_robot.py", line 171, in main
    run_simulator(sim, scene)
  File "/home/stardust/simulation/IsaacLab/source/standalone/tutorials/04_sensors/add_sensors_on_robot.py", line 125, in run_simulator
    scene.reset()
  File "/home/stardust/simulation/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py", line 276, in reset
    sensor.reset(env_ids)
  File "/home/stardust/simulation/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 342, in reset
    raise RuntimeError(
RuntimeError: Camera could not be initialized. Please ensure --enable_cameras is used to enable rendering.
2024-06-25 09:07:21 [27,476ms] [Warning] [carb] [Plugin: omni.sensors.tiled.plugin] Module /home/stardust/.local/share/ov/pkg/isaac-sim-4.0.0/extscache/omni.sensors.tiled-0.0.3+106.0.0.lx64.r/bin/libomni.sensors.tiled.plugin.so remained loaded after unload request
2024-06-25 09:07:21 [27,480ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/stardust/.local/share/ov/pkg/isaac-sim-4.0.0/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2024-06-25 09:07:21 [27,481ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,481ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,481ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,481ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,481ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,481ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,481ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,481ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,481ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,482ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,482ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-25 09:07:21 [27,482ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-06-25 09:07:21 [27,667ms] [Warning] [carb] Client omni.syntheticdata.plugin Failed to acquire interface [omni::graph::core::INode v4.10] while unloading all plugins

Mayankm96, 2024-06-25:
Can you please follow the bug report template?

The error says to pass `--enable_cameras` flag. Did you do that?

ChengK321, 2024-06-25:
> Can you please follow the bug report template?
> 
> The error says to pass `--enable_cameras` flag. Did you do that?

I tried running the following command in the terminal:  
python source/standalone/tutorials/04_sensors/add_sensors_on_robot.py --enable_cameras
This time the isaac sim rendered the interface, but the interface was weird, and the terminal still reported an error.
[image]
[image]

ChengK321, 2024-06-27:
The main cause of this problem is the following error: 

> _Traceback (most recent call last):
>   File "/home/stardust/.local/share/ov/pkg/isaac-sim-4.0.0/kit/exts/omni.syntheticdata/omni/syntheticdata/scripts/SyntheticData.py", line 1622, in [...]

The solution is to find this directory IsaacLab/_isaac_sim/kit/exts/omni.kit.window.viewport/omni/kit/viewport_legacy/, and rename the *.so file to _viewport_legacy.cpython_310_x86_64_linux_gnu.so

closed 2024-06-28 by ChengK321