isaac-sim/IsaacLab #569: [Question] Import Objaverse glb file into Isaac Sim

supersglzc, 2024-06-26:
### Question

Hi, I would like to import some objects from Objaverse. They are glb files and I want to convert them to usd files. Is there any exist converter for this? 

I found the asset converter https://docs.omniverse.nvidia.com/extensions/latest/ext_asset-converter.html in Omniverse but I got no such module error. 

Thanks!

Mayankm96, 2024-06-26:
We have some instructions on importing new assets: https://isaac-sim.github.io/IsaacLab/source/how-to/import_new_asset.html#using-mesh-importer

Hope that helps!

supersglzc, 2024-06-27:
Thanks for the reply! This importer does convert the mesh to usd. However, the collision mesh does not exist and it only has a geometry Xform, even though I select the collision approximation to convexDecomposition. 

I knew that the mesh itself has collider property, but when I import it into my environment, it gives me very weird collision.