isaac-sim/IsaacLab #570: [Bug Report] Can't start training. [omni.appwindow.plugin] Failed to acquire IWindowing interface

1g0rrr, 2024-06-26:
### Describe the bug
When I try to start any training in AWS cloud with IsaacLab I got error "[Error] [omni.appwindow.plugin] Failed to acquire IWindowing interface" and script stops.
What can be the reason of this?
 
[image]

### Steps to reproduce
Install IsaacSim with IsaacLab to to Amazon EC2 manually or via Automator
run 
"./isaaclab.sh -p source/standalone/tutorials/03_envs/create_quadruped_base_env.py"

- Isaac Sim Version: 4.0.0
- GPU: g4dn.x2large (Tesla T4)

Full log:
```
[IsaacLab]:/isaaclab$ ./isaaclab.sh -p source/standalone/tutorials/03_envs/create_quadruped_base_env.py
[INFO] Using python from: /isaac-sim/python.sh                                                   
[INFO][AppLauncher]: Loading experience file: /isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.kit
Starting kit application with the following args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/renderer/activeGpu=0', '--/physics/cudaDevice=0', '--portable', '--/app/window/hideUi=False', '--allow-root']
Passing the following args to the base kit application:  []
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/4.0/kit_20240626_021418.log
2024-06-26 02:14:18 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.074s] [ext: omni.kit.async_engine-0.0.0] startup
[0.662s] [ext: omni.stats-0.0.0] startup
[0.663s] [ext: omni.client-1.1.0] startup
[0.701s] [ext: omni.datastore-0.0.0] startup
[0.702s] [ext: omni.blobkey-1.1.0] startup
[0.702s] [ext: omni.ujitso.default-1.0.0] startup
[0.703s] [ext: omni.hsscclient-0.0.0] startup
[0.704s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.705s] [ext: omni.assets.plugins-0.0.0] startup
[0.707s] [ext: omni.gpu_foundation-0.0.0] startup
[0.721s] [ext: carb.windowing.plugins-1.0.0] startup
2024-06-26 02:14:19 [710ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-06-26 02:14:19 [710ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
[0.722s] [ext: omni.kit.renderer.init-0.0.0] startup
2024-06-26 02:14:19 [723ms] [Warning] [omni.platforminfo.plugin] failed to open the default display.  Can't verify X Server version.
2024-06-26 02:14:19 [876ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-06-26 02:14:19 [876ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.183.01    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | Tesla T4                         | Yes: 0 |     | 16630   MB | 10de      | 0          |
|     |                                  |        |     |            | 1eb8      | 8f287c4c.. |
|     |                                  |        |     |            | 0         |            |
|=============================================================================================|
| OS: 22.04.3 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.3, Kernel: 6.5.0-1020-aws
| Processor: Intel(R) Xeon(R) Platinum 8259CL CPU @ 2.50GHz | Cores: 4 | Logical: 8
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31714 | Free Memory: 29906
| Total Page/Swap (MB): 32767 | Free Page/Swap: 32767
|---------------------------------------------------------------------------------------------|
[1.877s] [ext: omni.kit.pipapi-0.0.0] startup
[1.879s] [ext: omni.kit.pip_archive-0.0.0] startup
[1.880s] [ext: omni.pip.compute-1.3.1] startup
[1.880s] [ext: omni.pip.cloud-1.1.3] startup
[1.885s] [ext: omni.isaac.core_archive-2.3.0] startup
[1.885s] [ext: omni.mtlx-0.1.0] startup
[1.886s] [ext: omni.usd.config-1.0.3] startup
[1.891s] [ext: omni.gpucompute.plugins-0.0.0] startup
[1.891s] [ext: omni.usd.libs-1.0.0] startup
[2.014s] [ext: omni.isaac.ml_archive-2.0.1] startup
[2.015s] [ext: omni.kit.loop-isaac-1.2.0] startup
[2.016s] [ext: omni.kit.test-0.0.0] startup
[2.079s] [ext: omni.kit.telemetry-0.5.0] startup
[2.112s] [ext: omni.appwindow-1.1.8] startup
2024-06-26 02:14:20 [2,102ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-06-26 02:14:20 [2,102ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-06-26 02:14:20 [2,103ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-06-26 02:14:20 [2,103ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-06-26 02:14:20 [2,104ms] [Error] [omni.appwindow.plugin] Failed to acquire IWindowing interface
[2.117s] [ext: omni.kit.renderer.core-1.0.1] startup
Fatal Python error: Segmentation fault

Current thread 0x00007b058d58eb80 (most recent call first):
  File "/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 350 in _start_app
  File "/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 197 in __init__
  File "/isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 536 in _create_app
  File "/isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 117 in __init__
  File "/isaaclab/source/standalone/tutorials/03_envs/create_quadruped_base_env.py", line 37 in <module>

Extension modules: psutil._psutil_linux, psutil._psutil_posix (total: 2)
/isaac-sim/python.sh: line 41:   233 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python

```

swstbecrpgmail, 2024-06-28:
You have to add argument --livestream 1 or --headless, for example:
"./isaaclab.sh -p source/standalone/tutorials/03_envs/create_quadruped_base_env.py -- livestream 1"
I have never used AWS before, and used Omniverse Streaming Client to visualize.

1g0rrr, 2024-06-30:
> You have to add argument --livestream 1 or --headless, for example: "./isaaclab.sh -p source/standalone/tutorials/03_envs/create_quadruped_base_env.py -- livestream 1" I have never used AWS before, and used Omniverse Streaming Client to visualize.

It helps me go further, thanks! 
Documentation is quite limited about this topic.
What is your workflow of using Isaac Lab?

swstbecrpgmail, 2024-07-01:
> It helps me go further, thanks! Documentation is quite limited about this topic. What is your workflow of using Isaac Lab?

Normally, you only need to run the script by adding arguments --livestream 1 or --headless, and visualize using omniverse streaming client, AWS and so on. The others are same with the documentation.

Mayankm96, 2024-07-02:
@hhansen-bdai Can we do something about this in the docs?

We specify a lot in the module-level docs but it seems the information is somewhat hidden from new users: https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.app.html#module-omni.isaac.lab.app

closed 2025-01-05 by RandomOakForest