isaac-sim/IsaacLab #575: [Question] Could it be possible to add Digit v3 as usd file?

xav-nal, 2024-06-27:
### Question

The robot Digit v4 is available in Isaac Sim and works with Isaac Lab. However, many research papers use the Digit robot v3, so it would be beneficial to have Digit v3 in Isaac Sim. Unfortunately, that is not currently available. Additionally, the Mujoco interpreter does not work well to convert a Digit v3 description file from XML to USD.

evanzijianhe, 2024-07-02:
Yes, please. We are also using DIgit v3

Mayankm96, 2024-07-04:
While we'd like to add support to this model, we are currently swamped with many requests.

You can create the asset yourself using the steps here: https://isaac-sim.github.io/IsaacLab/source/how-to/import_new_asset.html

For the digit-v3 robot, you'd need to add closed-loop linkages, but I am not sure if there are open-sourced models that have the correct kinematics.

closed 2024-09-26 by Dhoeller19