isaac-sim/IsaacLab #586: [Bug Report] Error while creating a singularity image from IsaacSim docker container using fakeroot

shafeef901, 2024-06-27:
Creating a singularity image from IsaacSim container using fakeroot fails (instructions followed as per the documentation). The following command causes the issue:
```bash
./docker/container.sh push
```
More specifically, the following line 353 causes the issue:
```bash
APPTAINER_NOHTTPS=1 apptainer build --sandbox --fakeroot isaac-lab-$container_profile.sif docker-daemon://isaac-lab-$container_profile:latest
```
Removing the `fakeroot` seems to fix the issue. However, I am not sure of the consequences of doing that.

### Steps to reproduce
```bash
./docker/container.sh push
```
### Logs and Error messages:
```
[INFO] Using container profile: base                                                                                                                                                                      
[INFO]: Building singularity with docker version: 24.0.7, and Apptainer version: 1.2.5.
newgidmap: Target 22044 is owned by a different user: uid:1002 pw_uid:1002 st_uid:1002, gid:999 pw_gid:1002 st_gid:999
ERROR  : 'newgidmap' execution failed. Check that 'newgidmap' is setuid root or has setcap cap_setgid+eip.
ERROR  : Error while waiting event for user namespace mappings: no event received
```

### System Info

- Commit: 915ebdd60de4bb8e270fe7e9f3807f5cd43c810e
- Isaac Sim Version: 4.0.0-rc.21+4.0.13872.3e3cb0c9.gl
- Docker Version: 24.0.7
- Apptainer Version: 1.2.5
- OS: Ubuntu 22.04
- GPU: RTX 3060
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Confirm if fakeroot is required

pascal-roth, 2024-09-11:
Hi,
according to the [documentation](https://apptainer.org/docs/user/1.0/fakeroot.html) `fakeroot` allows a user to have root privileges inside the container that does not have root privileges on the machine where the docker is build. In the case, that you have root privileges on your host machine, there should be no consequences. 
I am not sure where the error originates from, I couldn't reproduce it on my machine. Did you experience any issue when working with the singularity image?

RandomOakForest, 2025-01-05:
Please reopen the issue if we can provide further help. Thanks.

closed 2025-01-05 by RandomOakForest