isaac-sim/IsaacLab #589: [Question] 2024-06-28 04:21:46  [Error] [carb] [Plugin: omni.sensors.nv.lidar.ext.plugin]

936769651, 2024-06-28:
### Question

 python source\standalone\tutorials\00_sim\create_empty.py

I Can open the window normally, but there is an error, how to solve it?thank you.

2024-06-28 04:21:46  [Error] [carb] [Plugin: omni.sensors.nv.lidar.ext.plugin] Dependency: [omni::sensors::lidar::IGenericModelOutputIOFactory v0.1] failed to be resolved.

[image]

Logan1904, 2024-07-12:
Have the same problem here.

Followed this installation guide for Windows: https://isaac-sim.github.io/IsaacLab/source/setup/installation/binaries_installation.html

Songdada3, 2024-07-16:
I also encountered the same problem. Is there any solution?
{2024-07-16 01:00:02  [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-07-16 01:00:02  [Error] [carb] [Plugin: omni.sensors.nv.lidar.ext.plugin] Dependency: [omni::sensors::lidar::IGenericModelOutputIOFactory v0.1] failed to be resolved.
2024-07-16 01:00:02  [Warning] [omni.kit.profiler.window] remove _SpanInstance.__lt__ and use insort 'key' arg instead
}

bob-021206, 2024-07-19:
Following the offical doc from NVIDIA, if you are starting isaac sim on Windows, this error could be ignored. Cited from Errors No.13: https://docs.omniverse.nvidia.com/isaacsim/latest/known_issues.html

kellyguo11, 2024-09-27:
Yes, this error can be safely ignored.

closed 2024-09-27 by kellyguo11