isaac-sim/IsaacLab #604: [Question] Problems with a high number of environments when using Cameras

ArneKlages4444, 2024-06-30:
We encounter issues with a high number of environments when using Cameras. Although the memory problem with cameras is now fixed in Isaac Lab, we cannot scale up the amount of environments as we like. 

When using two graphic cards, we could achieve a parallelization of 475 environments in the test setup. When going up to 476, the setup sometimes works, or we get one of two errors (see below, errors 1 and 2). With more than 476 environments (tested up to 512), the tests always fail with error 1. When using only one graphics card, we could not even achieve 475 environments and got error 3. The memory usage was tracked while we were performing the experiments, and there was always plenty of VRAM left (see memory usage with 475). We did these tests on a Linux machine with TITAN RTX GPUs and reproduced them on our HPC with L40S GPUs. On both systems, the breaking points were the same, despite the more powerful L40S GPUs. 

For our test, we simply put a camera into the `FrankaCubeLift` environment and added the flattened image to the observations (code below).

Is there anything we do wrong in our setups, or is this a known issue with cameras? When using the `TiledCamera` camera, we could use 4096 parallel environments, but since it does not provide the desired image output (https://github.com/isaac-sim/IsaacLab/issues/493) it is not usable for us.

**System Info:**

System 1:

- For testing, we use the Docker container with X11 forwarding disabled.
- IsaacLab version (last commit): https://github.com/isaac-sim/IsaacLab/pull/497
- Isaac Sim Version: 4.0.0
- OS: Ubuntu 22.04.3 LTS
- GPU: 2 * NVIDIA TITAN RTX (24 GiB)
- CUDA Toolkit 11.5, Driver 12.0
- GPU Driver: 525.147.05

System 2 (HPC):

- IsaacLab version (last commit): https://github.com/isaac-sim/IsaacLab/pull/600
- Isaac Sim Version: 4.0.0
- GPU: 2 * L40S (48 GiB)
- CUDA Version: 12.2
- GPU Driver Version: 535.161.08

memory usage with 475:

+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.147.05   Driver Version: 525.147.05   CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA TITAN RTX    Off  | 00000000:17:00.0 Off |                  N/A |
| 41%   35C    P2    78W / 280W |   9934MiB / 24576MiB |     42%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  NVIDIA TITAN RTX    Off  | 00000000:B3:00.0  On |                  N/A |
| 40%   37C    P2    86W / 280W |   8375MiB / 24576MiB |     28%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+

Error 1 (two GPUs and num_envs > 476):

2024-06-24 10:54:07 [399,118ms] [Warning] [carb] Client carb.scenerenderer-rtx.plugin has acquired [carb::settings::ISettings v1.0] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
2024-06-24 10:57:26 [598,443ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_INITIALIZATION_FAILED
2024-06-24 10:57:26 [598,443ms] [Error] [carb.graphics-vulkan.plugin] vkCreateSemaphore failed.
2024-06-24 10:57:26 [598,443ms] [Error] [gpu.foundation.plugin] createSemaphore failed for interop semaphore
2024-06-24 10:57:26 [598,443ms] [Error] [gpu.foundation.plugin] Failed to create CUDA interop semaphore in command list "Render graph command list (Render queue 0, device 1, frame submission index 0)".
2024-06-24 10:57:56 [628,444ms] [Error] [gpu.foundation.plugin] Submission barrier "CUDA event dependency in MGPU resource" timeout in command list "Render graph command list (Render queue 0, device 0, frame submission index 0)"!
2024-06-24 10:57:56 [628,444ms] [Error] [gpu.foundation.plugin] Wait count: 1, Received signals: 0: 
2024-06-24 10:58:26 [658,466ms] [Error] [gpu.foundation.plugin] Submission barrier "CUDA event dependency in MGPU resource" timeout in command list "Render graph command list (Render queue 0, device 0, frame submission index 0)"!
2024-06-24 10:58:26 [658,466ms] [Error] [gpu.foundation.plugin] Wait count: 1, Received signals: 0:

Error 2 (two GPUs and sometimes at num_envs = 476):

Traceback (most recent call last):
  File "/workspace/isaaclab/test_fr.py", line 60, in <module>
    env = ManagerBasedRLEnv(cfg=env_cfg, render_mode="rgb_array")
[... 93 lines cut ...]
        self.observations.policy.image = ObsTerm(func=camera_img)

env_cfg = FrankaObjectLiftEnvCfg()
env_cfg.scene.num_envs = args_cli.num_envs
env = ManagerBasedRLEnv(cfg=env_cfg, render_mode="rgb_array")

def run_nvidia_smi():
    try:
        output = subprocess.check_output(['nvidia-smi'], text=True)
        print(output)
    except subprocess.CalledProcessError as e:
        print(f"Failed to run nvidia-smi: {e}")
    except FileNotFoundError:
        print("nvidia-smi command not found. Ensure you have the NVIDIA drivers installed.")

run_nvidia_smi()
for i in range(100):
    a = torch.randn_like(env.action_manager.action)
    _ = env.step(a)
    if i % 10 == 0:
        print("iteration:", i)
        run_nvidia_smi()
```

Console command: `./isaaclab.sh -p test_fr.py --headless --enable_cameras --num_envs=476`

btx0424, 2024-07-04:
Just another user here. 

Maybe give `TiledCameraCfg` a try. I am able to get >512 cameras running on my laptop RTX 4070.

ArneKlages4444, 2024-07-04:
Yes, the tiled camera achieves the necessary number of environments, but as I already mentioned, the image it outputs is right now not comparable to the normal camera. Using the `TiledCamera` would be our preferred option because of its much better performance, but in our projects, we need realistic images.

"When using the `TiledCamera` camera, we could use 4096 parallel environments, but since it does not provide the desired image output (https://github.com/isaac-sim/IsaacLab/issues/493) it is not usable for us."

kisisjrlly, 2024-07-17:
In a large number of camera simulation scenarios, I also encountered a similar error, is your problem solved?

ArneKlages4444, 2024-07-17:
No, sadly, it's not solved yet. The only way to increase the number of environments that I found is to add additional GPUs. But the increase in parallel environments is not linear to the amount of separate GPUs. So I would not use too many, especially because only a fraction of the GPU VRAM capacity is used.

johnnylu305, 2024-09-17:
I got same warning and my simulation freezes with 256 environments on my A6000 machine.

glvov-bdai, 2024-09-27:
Tiled rendering RGB has been improved in the 1.2 release, which fixes https://github.com/isaac-sim/IsaacLab/issues/493

I am able to simulate 2048 low resolution tiled cameras on a laptop 3080.

We are keeping an eye on https://github.com/isaac-sim/IsaacLab/issues/1031

You can try many cameras with your task environment  https://isaac-sim.github.io/IsaacLab/source/how-to/estimate_how_many_cameras_can_run.html . If you still experience this issue with tiled rendering let us know!

closed 2024-09-27 by glvov-bdai