isaac-sim/IsaacLab #611: Cannot download file from nucleus server: ret: (Result.ERROR_CONNECTION, '', <omni.client.Content object at 0x7f5c5bfb36f0>)

swstbecrpgmail, 2024-07-01:
I couldn't run "./orbit.sh -p source/standalone/demos/quadrupeds.py --livestream 1" correctly in docker container.

It first report that
[image]
[image]

and I find that the reason why I can't download the file is probably that it can't connect to nucleus server.
[image]

Additionally, I can successfully download the file using wget.
[image]

swstbecrpgmail, 2024-07-01:
> I couldn't run "./orbit.sh -p source/standalone/demos/quadrupeds.py --livestream 1" correctly in docker container.
>
> It first report that [...]

I use docker to run the program and it seems that the program trys to download files from nucleus server, while the Isaac Sim Docker container does not include Nucleus and will access assets directly from the Cloud by default.
I wonder whether the bug is related to this.

Mayankm96, 2024-07-02:
Is the docker / local machine connected to the Internet? Which commit tag of IsaacLab are you using?

It seems you are using an older version of the framework with Isaac Sim 2023.1.1 which is hard for us to check against.

RandomOakForest, 2025-01-05:
Our apologies we couldn't follow up in a timely manner. Please reopen the issue if you still see this using Isaac Sim 4.2 and Isaac Lab 1.4.0. Thank you.

closed 2025-01-05 by RandomOakForest