isaac-sim/IsaacLab #622: [Question] How to change USD visual color?

ahanjaya, 2024-07-03:
### Question

How do we change the visual_material color of the robot? I added visual material to the spawn in ArticulationCfg, the color of the robot doesn't change and causes some environments are not working

```
        spawn=sim_utils.UsdFileCfg(
        usd_path=f"{ISAACLAB_NUCLEUS_DIR}/Robots/Unitree/A1/a1.usd",
        activate_contact_sensors=True,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            retain_accelerations=False,
            linear_damping=0.0,
            angular_damping=0.0,
            max_linear_velocity=1000.0,
            max_angular_velocity=1000.0,
            max_depenetration_velocity=1.0,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=False, solver_position_iteration_count=4, solver_velocity_iteration_count=0
        ),
       visual_material = sim_utils.PreviewSurfaceCfg(
        diffuse_color=(random.random(), random.random(), random.random()), metallic=0.5
       ),
```

Mayankm96, 2024-07-03:
Since the asset is instanceable, you cannot change the properties of its visual meshes (this includes the material assigned to them). You will first need to make the asset uninstanceable.

For more information: https://openusd.org/dev/api/_usd__page__scenegraph_instancing.html

You can find the following function useful: https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sim.html#omni.isaac.lab.sim.utils.make_uninstanceable

ahanjaya, 2024-07-04:
Is this also the same reason that the asset is instanceable, 
```
[Warning] [omni.isaac.lab.sim.utils] Could not perform 'modify_collision_properties' on any prims under: '/World/envs/env_0/Animation'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.

```

How to `make_uninstanceable` I passed the prims_path, but causes different error

Thanks

Mayankm96, 2024-07-05:
Yes, that's the reason why that doesn't work. What error did you get? Can you provide more details?

ahanjaya, 2024-07-05:
I tried to make_uninstanceable with this commands:

```
for prim_path in sim_utils.find_matching_prim_paths("/World/envs/env_.*/Animation/.*/collisions"):
       print(prim_path)
       sim_utils.make_uninstanceable(prim_path)
```

and got following error:

```
/World/envs/env_0/Animation/trunk/collisions
2024-07-05 07:17:27 [11,037ms] [Warning] [omni.physx.tensors.plugin] prim '/World/envs/env_0/Animation/trunk/collisions/mesh_0' was deleted while being used by a shape in a tensor view class. The physics.tensors simulationView was invalidated.
2024-07-05 07:17:27 [11,039ms] [Error] [omni.physx.plugin] PhysX error: PxRigidActor::detachShape: shape is not attached to this actor!, FILE /builds/omniverse/physics/physx/source/physx/src/NpRigidActorTemplate.h, LINE 257
2024-07-05 07:17:27 [11,039ms] [Error] [omni.physx.plugin] PhysX error: PxRigidActor::detachShape: shape is not attached to this actor!, FILE /builds/omniverse/physics/physx/source/physx/src/NpRigidActorTemplate.h, LINE 257
2024-07-05 07:17:27 [11,039ms] [Error] [omni.physx.plugin] PhysX error: PxRigidActor::detachShape: shape is not attached to this actor!, FILE /builds/omniverse/physics/physx/source/physx/src/NpRigidActorTemplate.h, LINE 257
2024-07-05 07:17:27 [11,039ms] [Error] [omni.physx.plugin] PhysX error: PxRigidActor::detachShape: shape is not attached to this actor!, FILE /builds/omniverse/physics/physx/source/physx/src/NpRigidActorTemplate.h, LINE 257
2024-07-05 07:17:27 [11,039ms] [Error] [omni.physx.plugin] PhysX error: PxRigidActor::detachShape: shape is not attached to this actor!, FILE /builds/omniverse/physics/physx/source/physx/src/NpRigidActorTemplate.h, LINE 257
2024-07-05 07:17:27 [11,039ms] [Error] [omni.physx.plugin] PhysX error: PxRigidActor::detachShape: shape is not attached to this actor!, FILE /builds/omniverse/physics/physx/source/physx/src/NpRigidActorTemplate.h, LINE 257
2024-07-05 07:17:27 [11,040ms] [Error] [omni.physx.plugin] PhysX error: PxRigidActor::detachShape: shape is not attached to this actor!, FILE /builds/omniverse/physics/physx/source/physx/src/NpRigidActorTemplate.h, LINE 257
```

swstbecrpgmail, 2024-10-17:
UNITREE_A1_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path=f"{ISAACLAB_NUCLEUS_DIR}/Robots/Unitree/A1/a1.usd",
        activate_contact_sensors=True,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            retain_accelerations=False,
            linear_damping=0.0,
            angular_damping=0.0,
            max_linear_velocity=1000.0,
            max_angular_velocity=1000.0,
            max_depenetration_velocity=1.0,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=False, solver_position_iteration_count=4, solver_velocity_iteration_count=0
        ),
        **visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 0.0, 1.0))**
    ),
I added "**visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 0.0, 1.0))**" in UNITREE_A1_CFG, but the color is still black.

RandomOakForest, 2025-01-05:
Our apologies we couldn't follow up in a timely manner. Please reopen the issue if you still see this using Isaac Sim 4.2 and Isaac Lab 1.4.0. Thank you.

closed 2025-01-05 by RandomOakForest

el-refai, 2025-04-22:
> Since the asset is instanceable, you cannot change the properties of its visual meshes (this includes the material assigned to them). You will first need to make the asset uninstanceable.
>
> For more [...]

Once I have made the asset uninstanceable how can I change the visual material?