isaac-sim/IsaacLab #629: [Bug Report] Isaac Lab PIP Installation and Vscode Setup

Mayankm96, 2024-07-03:
### Discussed in https://github.com/isaac-sim/IsaacLab/discussions/620

<div type='discussions-op-text'>

<sup>Originally posted by **evanzijianhe** July  2, 2024</sup>
I installed the Isaac Sim by using pip install method, and also installed the Isaac Lab following the instruction. I was able to run the testing code to verify the installation. However, I have some questions regarding Vscode setup:

1. Do we still need to run the `setup_python_env` in the IsaacLab and IsaacLabExtensionTemplate if I was using pip install? I tried, it was impossible to properly execute it, because `_isaac_sim` symbolic link was missed. Does it mean we still need to install the IsaacSim using omniuniverse-launch and setup path for `_isaac_sim`? 
2. Can we install IsaacSim use both pip and pre-built binaries methods? 
3. When I import functions from omni.isaac.lab class, those function are unknown( but program can be executed). Is this normal? 
[image]

Can someone helped to answer those questions? Thank you very much!</div>

evanzijianhe, 2024-07-04:
Hi Mayank. Thanks for your helps. I have tested this fix. There was no `.vscode` fold in the isaacsim fold in anaconda, so I could not find `anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/.vscode/settings.json`.

However, I found that the `settings.json` file exits in the `isaacsim/extscache/omni.kit.window.extensions/ext_template/vscode`, so I manually created a `.vscode` fold in the root of isaacsim folder and copy the `settings.json` to here. Then, everything is fine.

Mayankm96, 2024-07-04:
@evanzijianhe PR #631 should solve the issue here.

closed 2024-07-29 by Mayankm96

toenails6, 2024-10-13:
I don't think this has been fixed yet, not for windows 11 isaac sim conda pip install at least. 
[image]

I also tried the quoted method, does not work either, just says 'Setting up vscode settings' and then nothing. But, relevant packages still show with warnings to cannot find relevant modules in vscode. 

> Hi Mayank. Thanks for your helps. I have tested this fix. There was no `.vscode` fold in the isaacsim fold in anaconda, so I could not find [...]

toenails6, 2024-10-13:
I tried installing using binaries by the way, also shows warnings in vscode.

AurelArnold, 2025-03-27:
I think I'm still encountering a similar issue described here.

I installed Isaac Sim from pip and then Isaac Lab from source code, following the instructions. When running `./isaaclab.sh --install` everything seems to work fine besides a warning at the end:
`[WARN] Could not find Isaac Sim VSCode settings: /home/aurel/anaconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/.vscode/settings.json.
	This will result in missing 'python.analysis.extraPaths' in the VSCode
	settings, which limits the functionality of the Python language server.
	However, it does not affect the functionality of the Isaac Lab project.
	We are working on a fix for this issue with the Isaac Sim team.
`
This doesn't disturb the installation and running scripts afterward works.

The problem I'm facing arises, when I create a project using the template generator and try to set up IDE. When running the VSCode Task, 'setup_pyhton_env' and leaving the path to Isaac Sim blank (as recommended for pip installations), I get the following error:
` *  Executing task: /python.sh /home/aurel/git/isaaclab_serenity/.vscode/tools/setup_vscode.py --isaac_path  

/usr/bin/bash: line 1: /python.sh: No such file or directory

 *  The terminal process "/usr/bin/bash '-c', '/python.sh /home/aurel/git/isaaclab_serenity/.vscode/tools/setup_vscode.py --isaac_path '" failed to launch (exit code: 127). 
 *  Terminal will be reused by tasks, press any key to close it. `

I would appreciate it if someone could help me, thanks!