isaac-sim/IsaacLab #632: [Proposal] Allow customizable device["product_string"] lookup in the Se2SpaceMouse and Se3SpaceMouse initialization

MandiZhao, 2024-07-03:
### Proposal

When initializing a `Se2SpaceMouse` or `Se3SpaceMouse` device, allow custom lookup product_strings. Also, sometimes `hid` detects device is already open even when the device is not actually used by other processes, so _maybe_ the runtime error here can be allowed.

For example, change the `self._find_device(self)` ([here](https://github.com/isaac-sim/IsaacLab/blob/1c4ea78544713c75b0a0ca2227c23213007dea0a/source/extensions/omni.isaac.lab/omni/isaac/lab/devices/spacemouse/se3_spacemouse.py#L122)) to: 

```
    def _find_device(self, product_string_lookup="SpaceMouse Compact"):
        """Find the device connected to computer."""
        found = False
        # implement a timeout for device search
        for _ in range(5):
            for device in hid.enumerate():
                if device["product_string"] == product_string_lookup:
                    # set found flag
                    found = True
                    vendor_id = device["vendor_id"]
                    product_id = device["product_id"]
                    # connect to the device
                    try:
                        self._device.open(vendor_id, product_id)
                    except RuntimeError:
                        print("device already opened")
            # check if device found
            if not found:
                time.sleep(1.0)
            else:
                break
        # no device found: return false
        if not found:
            raise OSError("No device found by SpaceMouse. Is the device connected?")
```
### Motivation
Current implementation assumes a fixed product_string `device["product_string"] == "SpaceMouse Compact"`
But other devices have different production strings, e.g. https://3dconnexion.com/us/ space mouse has `"3Dconnexion Universal Receiver"` as product_string

Mayankm96, 2024-07-03:
We can simplify our operations by switching to [pyspacemouse](https://pypi.org/project/pyspacemouse/#description). Should not be a lot of work to do that. If I recall correctly, we had a version of this.

@jsmith-bdai or @hhansen-bdai do you still have this code on your end that we can revive?

jsmith-bdai, 2024-08-10:
> We can simplify our operations by switching to [pyspacemouse](https://pypi.org/project/pyspacemouse/#description). Should not be a lot of work to do that. If I recall correctly, we had a version of this.
> 
> @jsmith-bdai or @hhansen-bdai do you still have this code on your end that we can revive?

Sorry I somehow missed this - we do have code to do this internally at the AI institute so can add it in post-release

SevenFo, 2025-03-20:
hi, how's the work going?

jsmith-bdai, 2025-04-06:
Hi @SevenFo and @MandiZhao, PR is up [here](https://github.com/isaac-sim/IsaacLab/pull/2263) to base our spacemouse device driver off of pyspacemouse. Feel free to try it out!

YongcanZhou, 2025-07-14:
> Hi [@SevenFo](https://github.com/SevenFo) and [@MandiZhao](https://github.com/MandiZhao), PR is up [here](https://github.com/isaac-sim/IsaacLab/pull/2263) to base our spacemouse device driver off of pyspacemouse. Feel free to try it out!

Hello, I tried to modify the file using this branch: https://github.com/isaac-sim/IsaacLab/pull/2263/files

1、scripts/environments/teleoperation/teleop_se3_agent.py
2、
source/isaaclab/isaaclab/devices/spacemouse/base_spacemouse.py
3、
source/isaaclab/isaaclab/devices/spacemouse/se2_spacemouse.py
4、source/isaaclab/isaaclab/devices/spacemouse/se3_spacemouse.py

Then pyspacemouse==1.1.3 and easyhid are installed.

Run./isaaclab.sh-pscripts/tools/record _ demos.py-task Isaac-stack-cube-franka-ik-rel-v0-teleop _ device space mouse-dataset _ file./datasets/dataset.hdf5,

Print with
```

> Creating window for environment.
> [INFO]: Completed setting up the environment ...
> 3Dconnexion Universal Receiver found
> 3Dconnexion Universal Receiver found
> 3Dconnexion Universal Receiver found
> 3Dconnexion Universal Receiver found
> 3Dconnexion Universal Receiver found
> 3Dconnexion Universal Receiver found
> 3Dconnexion Universal Receiver found
> 3Dconnexion Universal Receiver found
> 3Dconnexion Universal Receiver found
> ----------------------------------------------
> The SpaceMouse Controller connected:
> Manufacturer: 3Dconnexion
> Product: 3Dconnexion Universal Receiver
> ----------------------------------------------

```

But I use the space mouse, and the robot can't move. Excuse me, am I missing any steps? Thank you very much.