isaac-sim/IsaacLab #641: [Bug Report] Center of Mass cannot be set with GPU pipeline

AlanSunHR, 2024-07-04:
### Describe the bug

Cannot randomize the center of masses of bodies in an Articulation, with error "2024-07-04 09:45:53 [14,435ms] [Error] [omni.physx.tensors.plugin] Center of Mass cannot be set with GPU pipeline."

### Steps to reproduce

I wrote some codes to randomize the center of mass:
```
def randomize_body_coms(
    env: ManagerBasedEnv, env_ids: torch.Tensor | None, max_displacement: float, asset_cfg: SceneEntityCfg
):
    """Randomize the CoM of the bodies by adding a random value sampled from the given range.

    .. tip::
        This function uses CPU tensors to assign the CoM. It is recommended to use this function
        only during the initialization of the environment.
    """
    # extract the used quantities (to enable type-hinting)
    asset: Articulation = env.scene[asset_cfg.name]

    # resolve environment ids
    if env_ids is None:
        env_ids = torch.arange(env.scene.num_envs, device="cpu")
    else:
        env_ids = env_ids.cpu()

    # resolve body indices
    if asset_cfg.body_ids == slice(None):
        body_ids = torch.arange(asset.num_bodies, dtype=torch.int, device="cpu")
    else:
        body_ids = torch.tensor(asset_cfg.body_ids, dtype=torch.int, device="cpu")

    # get the current com of the bodies (num_assets, num_bodies)
    coms = asset.root_physx_view.get_coms().clone()[:, body_ids, :3]

    # Randomize the com in range -max displacement to max displacement
    coms += torch.rand_like(coms) * 2 * max_displacement - max_displacement

    # Set the new coms
    new_coms = asset.root_physx_view.get_coms().clone()
    new_coms[:, asset_cfg.body_ids, 0:3] = coms
    asset.root_physx_view.set_coms(new_coms, env_ids)
```

when the above function executes, it failed with error:
```
2024-07-04 09:45:53 [14,435ms] [Error] [omni.physx.tensors.plugin] Center of Mass cannot be set with GPU pipeline.
```

### System Info

Describe the characteristic of your environment:

- Commit: 1c4ea785
- Isaac Sim Version: 4.0.0
- OS: ubuntu 22.04
- GPU: RTX 4080
- CUDA: 12.3
- GPU Driver: 545.29.06

### Additional context

The randomization of center of mass at start up worked well in Orbit, but not in IsaacLab. Randomization of body masses at start up works in IsaacLab. I tried to write the randomization of CoM as similiar as possible to the mass, but it seems like a problem at the backend.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Milad-Rakhsha, 2024-07-05:
Yes I very recently fixed this for ISIM-1183

Milad-Rakhsha-NV, 2024-07-05:
I recently fixed this for ISIM-1183

closed 2024-07-05 by Milad-Rakhsha-NV