isaac-sim/IsaacLab #642: [Proposal] Dropping support for Isaac Sim 2023.1.1

Mayankm96, 2024-07-04:
### Proposal

This proposal advocates removing the backward compatibility of 2023.1.1 to focus on the API changes from Isaac Sim 4.0.

### Motivation

It has been a few months since Isaac SIm 4.0 is out. IsaacLab tries to maintain backward compatibility with Isaac Sim 2023.1.1.  However, our CI/CD pipelines and developers barely check that the framework is working thoroughly with 2023.1.1.

Moving to support Isaac Sim 4.0 and onwards will bring the following benefits:

* Upgrading the contact sensor class to report friction data without a lot of version-checking
* Upgrading articulation class to use only the PhysX articulation view and avoid parsing the rigid body view (which was very limiting for nested links)
* Simplifying the conda and vscode environment setup to use pip binaries directly

### Alternatives

We painstakingly maintain 2023.1.1 without any guarantees for users. But this makes the code hard to maintain.

### Additional Context

IsaacLab always tries to stay compatible with the last two releases of Isaac Sim. Since Isaac Sim 4.1 will be out soon, it makes sense to focus on Isaac Sim 4.0 and Isaac Sim 4.1.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

- [ ] Drop support of Isaac Sim 2023.1.1 to focus on Isaac Sim 4.0 onwards developments

KyleM73, 2024-07-05:
As sort of a meta comment I think there’s trade offs to be had between active development and a stable framework to attract users to. In the last year that I’ve been using orbit/isaac lab there’s been I want to say 2-3 major breaking changes that requires significant code rewrites as an end user. In each of those cases the reason for the breaking change was very well justified and resulted in a significantly better more capable framework. And, given the details provided here, I think dropping support for 2023.1.1 is another example of a well justified change. I think a question to keep in mind is just “at what point do we have a release that’s stable enough to support continuously long enough so the community can build on top of it”. I think right now it’s too early to make that kind of trade off, each release is bringing really significant features that are very worth a bit of refactoring as a user, just something to look out for as people/labs start to build more internal infrastructure on top of Isaac lab, especially given that we’re all sort of beholden to Isaac sim releases, it’s not like pybullet etc. where you can just install a binary and use offline forever.

Mayankm96, 2024-08-13:
This has been resolved with Isaac Lab release 1.1.

closed 2024-08-13 by Mayankm96