isaac-sim/IsaacLab #645: Cannot load mdl correctly

swstbecrpgmail, 2024-07-05:
When I run "./isaaclab.sh -p source/standalone/workflows/rsl_rl/play.py --task Isaac-Velocity-Rough-Unitree-Go1-v0 --num_envs 32", it report that it failed to open the needed mdl: 
[image]

Then I download the needed mdl and changed the directory of the visual material: 
[image]
But it reports  "failed to resolve usd asset identifier":
[image]

Additionally, I am using the latest of the isaaclab and isaac sim 4.0.

If anyone has potential solutions, feel free to leave comments. Thank you very much!

Mayankm96, 2024-07-05:
@KyleM73 This seems similar to the issue you mentioned in #551 

@swstbecrpgmail can you check that issue and see it is the same setup that's causing issues for you too?

swstbecrpgmail, 2024-07-05:
> follow the bug report template

Sorry, I have edited the issue fllowing the bug report template. Thank you very much!

KyleM73, 2024-07-05:
> @KyleM73 This seems similar to the issue you mentioned in #551
> 

I’m not sure that this is related to #551 as my bug never appears with loading the rough terrain, it only happens in the container in headless mode on flat terrain with the viewer tracking the asset root when recording videos to wandb. In every other configuration that I’ve tried it’s worked as expected. Unless maybe there was some change to how MDLs are resolved internally?  I’m not sure what causes the initial issue here but the secondary issue of the dark environment feels like it’s caused by the path to the file not being resolved correctly when the cfg is loaded in the env. 

As a tangentially related issue that I know has been brought up before, it would be really really helpful to have documentation about what materials are in the nucleus server. I’ve crawled it manually to get a bunch of MDLs but it would be nice to know what’s in there directly and it would definitely help with debugging as well (since you can print the path to the MDL after it’s been resolved, it would be helpful to have a reference for what the path *should* be). It would also give a lot more flexibility for users wanting to make custom envs. I’m pretty sure this issue has been raised elsewhere though.

swstbecrpgmail, 2024-07-06:
> ### Describe the bug
> When I run "./isaaclab.sh -p source/standalone/workflows/rsl_rl/play.py --task Isaac-Velocity-Rough-Unitree-Go1-v0 --num_envs 32", it report that it failed to open the needed mdl file, which might be caused by the unstable connection to nucleus server (I have confirmed that I have connected to nucleus server and this bug did not occur during the previous run). Then I download the needed mdl file, changed the reading path of the mdl file from url to the local path on the local computer, it reports that "Texture file referenced in the material parameter defaults wasn't resolved properly", and the terrain looks very dark in isaac sim.
> 
> ### Steps to reproduce
> Firstly, run "./isaaclab.sh -p source/standalone/workflows/rsl_rl/play.py --task Isaac-Velocity-Rough-Unitree-Go1-v0 --num_envs 32".(You need to have a trained policy beforehand.)
> 
> ```
> ./isaaclab.sh -p source/standalone/workflows/rsl_rl/play.py --task Isaac-Velocity-Rough-Unitree-Go1-v0 --num_envs 32
> ```
> 
> You may not meet the "failed to open the needed mdl file" error below at first, but this error occurs to me after I run this command several times. I think this error is related to nucleus server, so if you want to reproduce this error, you may need to stop the nucleus server. (This might not be an important problem because I use wget to download the needed mdl file afterwards.) And Here are the terminal logs:
> 
> ```
> [INFO] Using python from: /home/crp/anaconda3/envs/isaaclab/bin/python                                                                                                    
> [INFO][AppLauncher]: Loading experience file: /home/crp/workspace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.kit
> Starting kit application with the following args:  ['/home/crp/workspace/IsaacLab/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/crp/workspace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.kit', '--/app/tokens/exe-path=/home/crp/workspace/IsaacLab/_isaac_sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/crp/workspace/IsaacLab/_isaac_sim/exts', '--ext-folder', '/home/crp/workspace/IsaacLab/_isaac_sim/apps', '--/renderer/activeGpu=0', '--/physics/cudaDevice=0', '--portable', '--/app/window/hideUi=False']
> Passing the following args to the base kit application:  ['--task', 'Isaac-Velocity-Rough-Unitree-Go1-v0', '--num_envs', '32']
> Loading user config located at: '/home/crp/.local/share/ov/pkg/isaac-sim-4.0.0/kit/data/Kit/Isaac-Sim/4.0/user.config.json'
> [Info] [carb] Logging to file: /home/crp/.local/share/ov/pkg/isaac-sim-4.0.0/kit/logs/Kit/Isaac-Sim/4.0/kit_20240705_202414.log
> 2024-07-05 12:24:14 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
> [0.106s] [ext: omni.kit.async_engine-0.0.0] startup
> [0.407s] [ext: omni.stats-0.0.0] startup
> [0.408s] [ext: omni.client-1.1.0] startup
> [0.437s] [ext: omni.datastore-0.0.0] startup
> [0.438s] [ext: omni.blobkey-1.1.0] startup
> [0.438s] [ext: omni.ujitso.default-1.0.0] startup
> [0.439s] [ext: omni.hsscclient-0.0.0] startup
> [0.439s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
> [0.440s] [ext: omni.assets.plugins-0.0.0] startup
> [0.441s] [ext: omni.gpu_foundation-0.0.0] startup
> [0.451s] [ext: carb.windowing.plugins-1.0.0] startup
> [0.459s] [ext: omni.kit.renderer.init-0.0.0] startup
> MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0
> 
> 2024-07-05 12:24:14 [679ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Graphics (RKL GT1)
> 2024-07-05 12:24:14 [680ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Graphics (RKL GT1)
> 
> |---------------------------------------------------------------------------------------------|
> | Driver Version: 535.161.08    | Graphics API: Vulkan
> |=============================================================================================|
> | GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
> |     |                                  |        |     |            | Device-ID | UUID       |
> |     |                                  |        |     |            | Bus-ID    |            |
> |---------------------------------------------------------------------------------------------|
> | 0   | NVIDIA GeForce RTX 3060          | Yes: 0 |     | 12534   MB | 10de      | 0          |
> |     |                                  |        |     |            | 2503      | 31bc2b95.. |
> |     |                                  |        |     |            | 1         |            |
> |---------------------------------------------------------------------------------------------|
> | 1   | Intel(R) Graphics (RKL GT1)      |        |     | 23872   MB | 8086      | 0          |
> |     |                                  |        |     |            | 4c8a      | e6537608.. |
[... cut ...] lx8)
> 
> ### Checklist
> * [x]  I have checked that there is no similar issue in the repo (**required**)
> * [ ]  I have checked that the issue is not in running Isaac Sim itself and is related to the repo
> 
> ### Acceptance Criteria
> Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.
> 
> * [ ]  Criteria 1
> * [ ]  Criteria 2

I have kind of solved this problem by downloading the assets-pack following

RandomOakForest, 2025-01-05:
Our apologies we couldn't follow up in a timely manner. Please reopen the issue if you still see this using Isaac Sim 4.2 and Isaac Lab 1.4.0. Thank you.

closed 2025-01-05 by RandomOakForest