isaac-sim/IsaacLab #654: Issue with running existing scripts

arthur801031, 2024-07-07:
System Info:
```
OS: Ubuntu 22.04.4 LTS x86_64 
Kernel: 5.15.0-112-generic 
Uptime: 7 days, 4 hours, 13 mins 
Packages: 3044 (dpkg), 15 (snap) 
Shell: bash 5.1.16 
Resolution: 6144x3456 
DE: GNOME 42.9 
WM: Mutter 
WM Theme: Adwaita 
Theme: Yaru [GTK2/3] 
Icons: Yaru [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel i9-10940X (28) @ 4.600GHz 
GPU: NVIDIA TITAN RTX 
Memory: 15670MiB / 128484MiB 

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Aug_15_22:02:13_PDT_2023
Cuda compilation tools, release 12.2, V12.2.140
Build cuda_12.2.r12.2/compiler.33191640_0
```

I followed the installation guide at https://isaac-sim.github.io/IsaacLab/source/setup/installation/pip_installation.html to install IssacLab; however, I'm getting the following error from running existing scripts (e.g., `./isaaclab.sh -p source/standalone/environments/teleoperation/teleop_se3_agent.py --task Isaac-Lift-Cube-Franka-IK-Rel-v0 --num_envs 1 --device keyboard`). How do I fix it?

```
...
[4.568s] Simulation App Starting
[7.122s] app ready
[8.586s] Simulation App Startup Complete
Traceback (most recent call last):
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/standalone/environments/teleoperation/teleop_se3_agent.py", line 43, in <module>
    import omni.isaac.lab_tasks  # noqa: F401
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/__init__.py", line 25, in <module>
    from .utils import import_packages
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/__init__.py", line 9, in <module>
    from .parse_cfg import get_checkpoint_path, load_cfg_from_registry, parse_env_cfg
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/parse_cfg.py", line 16, in <module>
    from omni.isaac.lab.envs import ManagerBasedRLEnvCfg
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/__init__.py", line 38, in <module>
    from . import mdp, ui
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/__init__.py", line 18, in <module>
    from .actions import *  # noqa: F401, F403
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/actions/__init__.py", line 8, in <module>
    from .actions_cfg import *
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/actions/actions_cfg.py", line 9, in <module>
    from omni.isaac.lab.managers.action_manager import ActionTerm, ActionTermCfg
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/__init__.py", line 13, in <module>
    from .action_manager import ActionManager, ActionTerm
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/action_manager.py", line 22, in <module>
    from .manager_base import ManagerBase, ManagerTermBase
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/manager_base.py", line 19, in <module>
    from .manager_term_cfg import ManagerTermBaseCfg
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/manager_term_cfg.py", line 19, in <module>
    from .scene_entity_cfg import SceneEntityCfg
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/scene_entity_cfg.py", line 11, in <module>
    from omni.isaac.lab.scene import InteractiveScene
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/__init__.py", line 28, in <module>
    from .interactive_scene import InteractiveScene
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py", line 19, in <module>
    from omni.isaac.lab.sensors import ContactSensorCfg, FrameTransformerCfg, SensorBase, SensorBaseCfg
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/__init__.py", line 36, in <module>
    from .camera import *  # noqa: F401, F403
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/__init__.py", line 8, in <module>
    from .camera import Camera
  File "/mnt/4TB-1/4TB-1/link_to_ssd/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 13, in <module>
    from tensordict import TensorDict
  File "/home/arthurliu/anaconda3/envs/issaclab/lib/python3.10/site-packages/tensordict/__init__.py", line 6, in <module>
    from tensordict._lazy import LazyStackedTensorDict
  File "/home/arthurliu/anaconda3/envs/issaclab/lib/python3.10/site-packages/tensordict/_lazy.py", line 25, in <module>
    from tensordict._td import _SubTensorDict, _TensorDictKeysView, TensorDict
  File "/home/arthurliu/anaconda3/envs/issaclab/lib/python3.10/site-packages/tensordict/_td.py", line 83, in <module>
    from torch.jit._shape_functions import infer_size_impl
  File "/home/arthurliu/anaconda3/envs/issaclab/lib/python3.10/site-packages/torch/jit/_shape_functions.py", line 1213, in <module>
    add_shape_compute_mapping(
  File "/home/arthurliu/anaconda3/envs/issaclab/lib/python3.10/site-packages/torch/jit/_shape_functions.py", line 1202, in add_shape_compute_mapping
    shape_compute_graph_mapping[operator_schema] = process_func(func)
  File "/home/arthurliu/anaconda3/envs/issaclab/lib/python3.10/site-packages/torch/jit/_shape_functions.py", line 1189, in process_func
    torch._C._jit_pass_inline(scripted_func.graph)
AttributeError: 'function' object has no attribute 'graph'
```

Mayankm96, 2024-07-07:
Do other scripts work? Such as `quadrupeds.py`?

arthur801031, 2024-07-07:
The scripts under [Showroom](https://isaac-sim.github.io/IsaacLab/source/setup/sample.html#showroom) work; however, other scripts don't work.

arthur801031, 2024-07-10:
I've tried various fixes, but none of them worked. The only way to get IssacLab to work is to create a new user account and install IssacLab in that account. I think my other research projects somehow interfere with IssacLab.

closed 2024-07-10 by arthur801031