isaac-sim/IsaacLab #656: [Question] Will more sensors be supported in the future?

shaoxiang, 2024-07-08:
### Question
Thank you very much for providing such an excellent robot simulation and learning platform.
The robot sensors I am using in simulation are LiDAR and IMU. 
Previously, I used RTX lidar in isaac sim, which supports Ouster, Velodyne, etc. I need to migrate them to isaac lab now.
I noticed that Velodyne lidar has been added #616 , but it is implemented using RayCaster. The information contained in RayCasterData is far inferior to RTX Lidar. For example, without intensity information.
(As mentioned @Mayankm96  in #532 **I think it is better to move the implementation to use RTX-based sensor since that will capture the LiDAR noise better.** )
So, how to migrate RTX Lidar to isaac lab, and is there any official support plan?
(I am referring to this open-source work: [go2_omniverse](https://github.com/abizovnuralem/go2_omniverse)

jakethesnake420, 2024-08-07:
I am working on the RTX lidar support privately right now and im making good progress.

```
    lidar = RTXRayCasterCfg(
        prim_path="{ENV_REGEX_NS}/Robot/base/lidar",
        offset=RTXRayCasterCfg.OffsetCfg(pos=(0.0, 0.0, 20.0)),
        spawn=sim_utils.LidarCfg()
    )
```
[image]

shaoxiang, 2024-08-07:
Very good work, looking forward to your further updates, this is enough for many perceptions. 
But I am afraid this is not a real RTX lidar, I understand that this cannot obtain lidar intensity information. @jakethesnake420

jakethesnake420, 2024-08-07:
@shaoxiang 
I am pretty sure intensity information is there. You can find more information here https://docs.omniverse.nvidia.com/isaacsim/latest/features/sensors_simulation/isaac_sim_sensors_rtx_based_lidar/annotator_descriptions.html

You can test out my PR now. https://github.com/isaac-sim/IsaacLab/pull/802

shaoxiang, 2024-08-08:
Excellent work, I will try it out now. I really need the lidar information with intensity.

closed 2025-01-05 by RandomOakForest