isaac-sim/IsaacLab #659: [Question] Do I import and run occupancy_map correctly?

yuezhezhang, 2024-07-08:
### Motivation

I want to test the functionality of occupancy_map and run the example code as shown [here](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.occupancy_map/docs/index.html). The steps I do are as follows:

#### Step 1:
As you mentioned [here](https://github.com/isaac-sim/IsaacLab/issues/131), I run the simulator first to be able to load extensions.

#### Step 2: 
As someone mentioned [here](https://forums.developer.nvidia.com/t/modulenotfounderror-no-module-named-omni-isaac-core-why-this-error-appear/272840), since Isaac Sim does not contain modules from `exts`. I added the path of `omni.isaac.occupancy_map` to the system:
```
import sys
sys.path.insert(0,'/home/frankie/git/isaac/IsaacLab/_isaac_sim/exts/omni.isaac.occupancy_map')
```
#### Step 3:
I changed this line `from omni.isaac.occupancy_map import _occupancy_map` to `from omni.isaac.occupancy_map.bindings import _occupancy_map`.

#### Step 4:
I loaded a scene with a rigid body object to check the functionality.

### Results:
**My commit message** is [here](https://github.com/yuezhezhang/IsaacLab/commit/28bc6e85445f168e2f64ac2a783d080958c426d2). It seems there is no bug and the printed results make sense to me.

### Question:
I want to make sure if the steps are correct and more generally speaking, can step 1 and 2 be used to **load other extensions of omni**?

Mayankm96, 2024-07-08:
The other way to do the same is by "enabling" the extension and importing the module from it.

As an example: https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/converters/urdf_converter.py#L117-L120

yuezhezhang, 2024-07-09:
Thanks for the help! I am closing this issue.

closed 2024-07-09 by yuezhezhang

HaoLiuHust, 2025-08-12:
@yuezhezhang what version do you use, I have test it in Isaac Sim 4.5 and Lab 2.1/2.2，it not works