isaac-sim/IsaacLab #660: [Question] How to update the current pose of primitives to the stage

notFoundThisPerson, 2024-07-09:
### Question

I want to read the current pose of the primitives in the scene for path planning with cuRobo, but the data read from omni.usd.get_world_transform_matrix() seems always be the initial pose of the prim. How can I get the current data, i.e. update the pose of the prim to the stage?

notFoundThisPerson, 2024-07-10:
Ok, I find that just set the `use_fabric=False` in `SimulationCfg` solves this problem, this disables the reading of physics buffers directly, but it may slow down the simulation of scenes with a large number of primitives.

closed 2024-07-10 by notFoundThisPerson

Sanjay1911, 2025-05-07:
Hello @notFoundThisPerson !

Did you successfully manage to integrate cuRobo into your Isaac Lab training ?
I'm also working on a similar replanning problem and would be a great help if you managed to do it already and share a few tips.

You can also reach me @ sanjayyadav.tamilselvam@tuhh.de

Regards,
Sanjay