isaac-sim/IsaacLab #663: [Question] Clarification on the units of measurement for joint actuators

GiulioRomualdi, 2024-07-10:
Hi all, thanks for developing and maintaining this framework 🎉 

I have a question about the unit of measurement used as input in the joint actuators. Coming from Isaac Gym, I was assuming that the joints were expressed in radians. However, in [ the documentation](https://isaac-sim.github.io/IsaacLab/source/migration/migrating_from_isaacgymenvs.html), I noticed that, unlike Isaac Gym, Isaac Sim assumes that joint velocity limits, damping, and stiffness are expressed in deg/s and 1/deg.

On the other hand, in the [actuator's documentation](https://isaac-sim.github.io/IsaacLab/source/features/actuators.html#actuators), there is no mention of the unit used to express the general quantities. Could you please help me identify the correct measurements used in the interface of IsaacLab?

cc @giulero

Mayankm96, 2024-07-29:
Ah yes. Unfortunately, USD seems to follow the `degrees` convention while the physics engine uses `radians.` Because of this, wherever we have a USD operation, the units are in degrees, and other places, they are radians.

Makes sense to unify the two and always use radians.

maxkra15, 2026-09-15:
Thanks @GiulioRomualdi for highlighting the old documentation ambiguity.

The current actuator API documents SI units explicitly: velocity limits are `[m/s or rad/s, depending on joint type]`, effort limits are `[N or N·m, depending on joint type]`, stiffness is `[N/m or N·m/rad]`, and damping is `[N·s/m or N·m·s/rad]`. The definitions are [here](https://github.com/isaac-sim/IsaacLab/blob/develop/source/isaaclab/isaaclab/actuators/actuator_base_cfg.py). Please reopen if anything remains unclear on a current Isaac Lab and Isaac Sim 6.1 installation.

closed 2026-09-15 by maxkra15