isaac-sim/IsaacLab #665: [Bug Report] Joint friction randomization

ori-gadot, 2024-07-10:
Bug Description:
Randomizing joint friction is not working.

Steps to Reproduce:
Run the script: source/standalone/tutorials/03_envs/create_cartpole_base_env.py with the following changes:

Remove all events from the EventCfg class. Add the following event:

`    robot_joint_limits = EventTerm(
        func=mdp.randomize_joint_parameters,
        mode="reset",
        params={
            "asset_cfg": SceneEntityCfg("robot", joint_names=".*"),
            "friction_distribution_params": (0.0, 1.0),
            "operation": "abs",
            "distribution": "uniform",
        },
    )
`

2. Change the CartpoleSceneCfg initialization as follows:
`scene = CartpoleSceneCfg(num_envs=64, env_spacing=0.0)`

3. Modify the simulation loop to inject zero efforts and reset the pole joint to a 0.1 position:

`    while simulation_app.is_running():
        with torch.inference_mode():
            # reset
            if count % 300 == 0:
                count = 0
                robot = env.scene["robot"]
                joint_pos = robot.data.default_joint_pos.clone()
                joint_pos[:,1] = 0.1
                joint_vel = robot.data.default_joint_vel.clone()
                robot.write_joint_state_to_sim(joint_pos, joint_vel)
                env.reset()
                print("-" * 80)
                print("[INFO]: Resetting environment...")
            joint_efforts = torch.zeros_like(env.action_manager.action)
            # step the environment
            obs, _ = env.step(joint_efforts)
            # print current orientation of pole
            print("[Env 0]: Pole joint: ", obs["policy"][0][1].item())
            # update counter
            count += 1
`

These changes will result in all environments being located at the same origin and the pole falling downwards due to gravity. The issue is that all environments behave the same, indicating that the DoF friction randomization did not take place.

Further Investigation:

1. Replacing friction_distribution_params with armature_distribution_params results in clearly different outcomes for different actors. This indicates that the overall process is legitimate, the events get triggered, and the joint armature values are indeed changed.

2. Changing the initial value of the joint friction in the actuators variable of the ArticulationCfg in the CARTPOLE_CFG configuration (source/extensions/omni.isaac.lab_assets/omni/isaac/lab_assets/cartpole.py) alters the physical simulation results. This suggests that the simulation is affected by different joint friction values.

3. Setting the DoF friction values directly using root_physx_view.set_dof_friction_coefficients was not effective either. Additionally, reading the values using root_physx_view.get_dof_friction_coefficients shows the values as changed, but they do not affect the simulation as expected.

SantiDiazC, 2024-07-10:
Hi @ori-gadot,
I just noticed you set the env_spacing to be 0.0:
```python
scene = CartpoleSceneCfg(num_envs=64, env_spacing=0.0)
```

That may be the reason all the environments are located at the same origin, idk if that solves any of the other issues you have.
I hope it helps.

ori-gadot, 2024-07-10:
Hi @SantiDiazC ,
This is intentional. In some cases, identical environments with different origins produce slightly different simulation results. To eliminate this variability and examine the pure randomized physics properties of the robot, I deliberately set the environments' spacing to zero.

Mayankm96, 2024-07-10:
@ori-gadot Thank you for opening this issue. We have reported this issue to the PhysX team, and a fix for this will be available in the upcoming Isaac Sim patch-fix release this month.

RandomOakForest, 2025-01-05:
Our apologies we couldn't follow up in a timely manner. Please reopen the issue if you still see this using Isaac Sim 4.2 and Isaac Lab 1.4.0. Thank you.

closed 2025-01-05 by RandomOakForest