isaac-sim/IsaacLab #668: [Proposal] Move lab_tasks gym registry to standalone module instead of relying on recursive imports

davis-openai, 2024-07-10:
### Proposal

Instead of relying on the recursive import of import_packages() in lab_tasks extension make a registry module that imports environments and registers them. Users should build environments anywhere but would register them all in the same file rather than in various places throughout the extension.

### Motivation

By relying on recursive imports through the top level __init__.py it's not possible to import individual modules from the lab_tasks extension. For instance, if I want to extend an environment in my own extension I have to import the entire lab_tasks extension rather than the specific pieces I need. 

### Alternatives

- Add registry.py top level module that does the recursive import (and modify that logic) so we can move imports out of __init__.py

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

- [ ] Individual modules can be imported from the lab_tasks extension without importing all extension dependencies.
- [ ] Individual modules can be imported from the lab_tasks extension without registering all environments with gym.

animesh-garg, 2024-07-15:
this is a very useful suggestion. 
Thanks @davis-openai 

@Mayankm96 @Dhoeller19 This change in import structure would enable lower overhead in users extending existing envs. 
Thoughts?

StafaH, 2026-10-01:
After doing some investigations in #7966 and debating the pros and cons, we have decided not to move towards a registry.py paradigm. If there are specific use cases that would benefit from #7966 please let us know and we can revisit the issue

closed 2026-10-01 by StafaH