isaac-sim/IsaacLab #674: [Question] Creating Scenes for Isaac Sim

arhanjain, 2024-07-11:
### Question

What's the best practice for setting up a USD scene for learning? Say I have a USD that just represents a mesh, and I want it to have collisions, and also have some articulations on drawers and cabinets and such. What's the best Cfg to use to load it in, and how can I prep that USD file to be simulation-ready? Currently when I try loading it in as a RigidObject via Python, it spawns rotated, despite spawning normally when I import it from the GUI. What would I use to setup the joints and stuff for articulation as well? I've seen many resources for bringing simulation-ready USDs into IsaacLab, but not much on creating such simulation ready assets. Any help is appreciated, thanks!

XuHu0529, 2024-11-27:
Hi, I also have the same question. So have you found a good solution?

Thanks for your help!

arhanjain, 2024-11-27:
hey @XuHu0529 yeah ive figured this stuff out now, whats your specific questions

XuHu0529, 2024-11-28:
I wonder how can I add collisions, rigid-body and articulations for a given USD asset via Python in IsaacLab. Thank you very much!