isaac-sim/IsaacLab #675: [Question] Will ROS affect the overall program performance? Especially when subscribing a topic

shaoxiang, 2024-07-12:
### Question
How to effectively use ROS in Isaac Lab? Any related tutorials or help?
I develop it under the Windows system,
CPU: Intel(R) Core(TM) i7-14700HX  
GPU: RTX4060
ROS Version: ROS2 Humble
After adding ros2 rclpy to my `play.py` program, it took a very long time for isaac lab to start, at least five minutes or more.
Another interesting phenomenon is that once I listen to a topic message, the entire Isaac Sim FPS will be pulled very low, dropping from around 20Hz to 0.5Hz. Once I unsubscribe, Isaac Sim can restore a higher frame rate. The phenomenon can be seen in GIF images.
Hope to get any useful help.
[image]

ramzibou0735, 2024-07-14:
I'd love to know too, since I'm working on strictly Sim2Real projects, using ROS as the main robotics framework is essential for good Sim2Real translation. 
Tho i do think your hardware is on the thin side when it comes to RL, you should be aiming for 12gb in VRAM and 32gb RAM

shaoxiang, 2024-07-15:
I tested with a stronger RTX 4070super desktop graphics card and it still showed the same problem.

ramzibou0735, 2024-07-15:
does it translate to the same performance or just a similar drop-off, how many FPS do you get after switching to your desktop?

shaoxiang, 2024-07-15:
Just a similar drop-off, Isaac Sim FPS dropping from around 20Hz to 10Hz. 
My suggestion is not to use ROS Python in Isaac Lab. You can try OmniGraph Ros2 instead. 
Anyway, I'm still waiting for an official response here.

ramzibou0735, 2024-07-15:
That would be a bummer if we were forced to not use ROS python but yeah lets wait for an official response and hopefully its just a bug

shaoxiang, 2024-07-16:
What is very frustrating is that when I display the point cloud in RVIZ, the 4070super desktop isaac sim also drops to 0.5FPS.
[image]
In Isaac Lab, I have tried to use rclpy to publish ros messages in many ways, including `while simulation_app.is_running()` and
using `threading` like:
```
thread_lidar = threading.Thread(target=run_lidar, args=(base_node,)) 
thread_lidar.start()
thread_imu = threading.Thread(target=run_imu, args=(base_node,)) 
thread_imu.start()
```
Similar FPS drops occur.

ramzibou0735, 2024-07-16:
Saw on your other issue that you have m_1-4 prop listed under your robot, are you working on a quadcopter project/research by any chance? If yes would it be possible to get your email/wechat or any contact? Would love your input on some related issues.

shaoxiang, 2024-07-16:
Yes, I am engaged in research of autonomous drone systems, and the experiments also require sim2real. As you said, ros is very important in sim2real. I am also involved in data-driven robot development. We have an Isaac lab communication group, and welcome to join the QQ group **`723139415`** for communication.

ramzibou0735, 2024-07-16:
That is also the research area I'm engaged it, Thanks for the invitation ! but i think the QQ group id is wrong since it only directs me to a person's contact, no QQ group.

shaoxiang, 2024-07-16:
Strange, the QQ group number is correct, you can add my QQ **519517384** or WeChat **sino-diy**

shaoxiang, 2024-07-16:
I tried more methods, such as using rclpy timers, or using rclpy Rates in many thread.
Like In `thread = threading.Thread(target=rclpy.spin, args=(node, ), daemon=True)`
It's the same problem. 
As long as I subscribe to the LiDAR point_cloud2 topic, Isaac Sim will be severely degraded, but subscribe to IMU topic will not have such a big impact.
I will continue to check the point_cloud2 processing code, but I think the subscription topic will affect the isaac sim thread.

shaoxiang, 2024-07-16:
I found that most of the compute resources in the point cloud processing part are spent on:
`point_cloud = point_cloud2.create_cloud(point_cloud.header, fields, data_points)`
This method comes from:
 `from sensor_msgs_py import point_cloud2`
Next, I will try other methods to **package point_cloud2**. What is puzzling is that `sensor_msgs_py ` is normal in the other ros program.

ramzibou0735, 2024-07-16:
My only guess without running the experiment is that pointcloud in Isaac is significantly more computationally hungry than on gazebo due to the difference in underlying physics?
By the way i just added you on QQ a couple minutes ago, we can continue our discussion there.

shaoxiang, 2024-07-16:
Finally, I finally found the real reason that affects the performance of isaac sim. 
It is caused by `lidar_pub.publish(point_cloud)`. 
Since the **point_cloud** data is relatively large, **publish** takes up more resources. 
But why does this cause the running frequency of isaac sim to drop very low?  Data publishing runs in a timer or separate thread.
Looking forward to the official reply...

shaoxiang, 2024-08-08:
The good news is that there is no such problem on Ubuntu 22.04.
I tested the above codes on windows11 system, isaac sim4.1.0 and Isaac lab v1.1.0, and the problem still exists. 
The conclusion is that there is a problem with the ros message publish in the Windows system.
I contribute my simulation release test code as follows:
[https://github.com/shaoxiang/IsaacLab/blob/main/source/standalone/tutorials/00_sim/publish_pointcloud.py](url)

AntoineRichard, 2026-08-27:
@shaoxiang, that's an unusually thorough investigation — thank you for narrowing it down yourself.

Your findings place the cost in `rclpy` publishing large `PointCloud2` messages, with no repro on Ubuntu 22.04 — that's ROS 2 / DDS transport behaviour on Windows and WSL2, and Isaac Lab neither ships nor wraps `rclpy` in its step loop. The ROS 2 bridge lives in Isaac Sim, so https://github.com/isaac-sim/IsaacSim/issues is the better place to track it — worth attaching your DDS config and the Windows/WSL2 vs 22.04 comparison, which is the most useful part of this thread.

@mzahana, your OmniGraph camera-publisher case under WSL2 is the same bridge and belongs there too. Closing here — please reopen if something turns out to sit on the Isaac Lab side after all.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard