isaac-sim/IsaacLab #680: [bug] Could Not Launch a Second Isaac Standalone  Instance During Training

zoctipus, 2024-07-15:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

When I train a environment using rsl_rl, if I try to train a second one using train.py or evaluate a checkpoint using play.py
This will not be possible because
[image]

### Steps to reproduce

Launch a training script, then launch a play script.

### System Info

Ubuntu 20.04
Isaac Sim 

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.0
- OS: Ubuntu 20.04
- GPU: 4090
- CUDA: 12.5
- GPU Driver: 555

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2024-07-29:
This is a limitation of Isaac Sim. I'm unsure what prevents it from launching two processes, but we have reported this issue to the team. Hopefully, a fix will make its way soon.

AntoineRichard, 2026-08-27:
Thanks for flagging this, @zoctipus — running a second standalone instance alongside training is a reasonable thing to want.

As Mayankm96 noted in the thread, the constraint on concurrent standalone app instances comes from the Kit/Isaac Sim layer (app and shader-cache locking) rather than from Isaac Lab, so that's where it's tracked. There's no Isaac Lab-side change to make, and the report predates several Isaac Sim releases. Closing here — if a second instance still won't launch on a current Isaac Sim, reopen with the exact error and version and we'll get it routed again.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard