isaac-sim/IsaacLab #682: [Question] Device error when using Physics Based IMU Sensor in Isaac Lab

shaoxiang, 2024-07-15:
### Question
It is inevitable to use the rich resources of Isaac Sim in Isaac Lab, but when I was using IMU sensors, an error occurred.
Whether I use Python from IMU API `omni.isaac.sensor to import IMUSensor` `get_current_frame` or use `OmniGarph` IMU read node, The following errors always occur:
> **[Error] [omni.physx.tensors.plugin] Incompatible device of velocity tensor in function omni::physx::tensors::GpuRigidBodyView::getVelocities: expected device 0, received device -1**

I have carefully read the tutorial on using IMU sensors:
https://docs.omniverse.nvidia.com/isaacsim/latest/advanced_tutorials/tutorial_advanced_isaac_imu_sensor.html
The sensor data read is also in CUDA Tensor, this error is really strange.And the IMU data I can read is as follows. 
**Note that the ang_vel data is all 0.0:**
> imu_data: {‘time’: 6.690000057220459, ‘physics_step’: 670.0, ‘lin_acc’: tensor([6.0603, 4.2792, 6.4186], device=‘cuda:0’), ‘ang_vel’: tensor([0., 0., 0.], device=‘cuda:0’), ‘orientation’: tensor([-0.5061, 0.1487, 0.3882, 0.7556], device=‘cuda:0’)}

**The data is in pytorch tensor format and is in cuda: 0 device.**

shaoxiang, 2024-07-16:
I have found through multiple experiments that as long as an IMU sensor is added to USD and this resource is opened in Isaac Lab, this error will continue to occur.
[image]

Mayankm96, 2024-07-29:
The IMU Sensor in Isaac Sim supports only CPU simulation (for some reason) and not other backends (besides `numpy`). It is expected that those features may not work with Isaac Lab that mainly supports `torch` backend for GPU-based parallelization.

We have an MR for IMU sensor integration into Isaac Lab. Hope that helps solve the issue for you.

shaoxiang, 2024-07-30:
Thank you for your explanation. I am looking forward to it. In fact, I have been following this IMU development work for a long time and hope it will be released as soon as possible. #619

closed 2024-09-26 by Dhoeller19