isaac-sim/IsaacLab #686: [Bug Report] Contact Force readings depending on Collider Type

choijoshua, 2024-07-15:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I was experimenting with the get_contact_force_matrix() function in RigidPrimView, and I realized that the values that the function is outputting changes based on the collider the asset has on Isaac Sim. When I define the RigidPrimView object, I set the contact_filter_prim_path_expr to be itself, like this. 
rigidprimview = RigidPrimView(prim_path, reset_xform_properties=False, track_contact_forces=True, contact_filter_prim_paths_expr=[prim_path], masses=mass)

### Steps to reproduce

Print the get_contact_force_matrix() output of a rigidprimview object in an environment every timestep, and see how the value changes when the rigid object that the rigidprimview is referencing to has different collider types. 

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

Mayankm96, 2024-07-29:
This seems wrong. Why would you set the contact filter path to be the prim itself? The filtering happens against a prim w.r.t. other prims in the scene (not the same prim).

rthaker01, 2024-09-26:
Hi @choijoshua - Are you still seeking assistance with this issue?

closed 2025-01-05 by RandomOakForest