isaac-sim/IsaacLab #687: [Question] Error when learning RSL-RL after updating to the latest Isaac Lab in a conda environment

JeonHaneul, 2024-07-16:
### Question

I recently heard that Isaac Lab now supports multi-GPU training for skrl, so I updated Isaac Lab and my conda environment to the latest versions. However, when I try to train with RSL-RL, I get the following error: "ValueError: Reference at 'refs/heads/master' does not exist." I've tried reinstalling both the conda environment and Isaac Lab, but the problem persists. I also tested other RL libraries like SKRL and RL games, and they seem to work fine, with training proceeding without any issues. It's only RSL-RL that's giving this error. Can you suggest any solutions?

[image]

JeonHaneul, 2024-07-17:
> Can you share the output of the rsl-rl version?
> 
> ```
> ./isaaclab.sh -p -m pip show rsl_rl
> ```

Thank you for your response. It appears from the command you provided that the version of RSL-RL is 2.0.2.

[image]

Koo-bone-seok, 2024-09-04:
this is because git setting. if you have .git folder in home directory, erase the .git folder

andrschl, 2024-09-11:
I observed the same issue. Removing the .git folder works, but also removes git from my custom extension. Is there a solution without removing the .git folder? Thanks in advance.

RandomOakForest, 2025-01-05:
Our apologies we couldn't follow up in a timely manner. Please reopen the issue if you still see this using Isaac Sim 4.2 and Isaac Lab 1.4.0. Thank you.

Steven9Smith, 2025-01-05:
this is still an issue, i can't run any of the examples using rsl_sl with the same Value error

Steven9Smith, 2025-01-15:
I pulled the latest changes yesterday and tested it and it's working now. Thank you.

training cmd: isaaclab.sh -p -m torch.distributed.run --nproc_per_node=2 --nnodes=1 standalone/workflows/skrl/train.py --task Isaac-Velocity-Flat-Spot-v0 --num_envs 2048 --distributed --headless

play cmd: isaaclab.sh -p standalone/workflows/skrl/play.py --task Isaac-Velocity-Flat-Spot-Play-v0 --num_envs 1 --checkpoint logs\skrl\spot_flat\2025-01-14_20-01-16_ppo_torch\checkpoints\agent_480000.pt

closed 2025-01-22 by kellyguo11