isaac-sim/IsaacLab #688: [Bug Report] Docker X11 Forwarding error when repo cloned in external drive

j3soon, 2024-07-16:
### Describe the bug

If the `IsaacLab` repo is cloned in an external drive, X11 forwarding settings for docker will be ignored due to `yq` snap package permission errors.

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

```sh
# Assume that /mnt/projects mounts an external drive
cd /mnt/projects
git clone https://github.com/isaac-sim/IsaacLab
cd IsaacLab
# Follow the docker installation guides here: https://isaac-sim.github.io/IsaacLab/source/deployment/docker.html
# Set `__ISAACLAB_X11_FORWARDING_ENABLED: 1` in `docker/.container.yaml`
./docker/container.sh start
```

Observe the following error message (the script will skip the error and continue starting the container):

```
[INFO] Using container profile: base
[INFO] Building the docker image and starting the container isaac-lab-base in the background...
[INFO] Required 'yq' package could not be found. Would you like to install it via snap? (y/N)y
[sudo] password for johnson:
yq v4.44.2 from Mike Farah (mikefarah) installed
Error: open /mnt/projects/IsaacLab/docker/.container.yaml: permission denied
[INFO] X11 Forwarding is configured as  in .container.yaml
[INFO] To enable X11 forwarding, set __ISAACLAB_X11_FORWARDING_ENABLED=1 in .container.yaml
[+] Building 1.8s (15/15) FINISHED
```

Attempting to launch any GUI program inside the container will cause segfault due to not being able to launch the app window.

```sh
./docker/container.sh enter
# in the container
/isaac-sim/runapp.sh
```

### System Info

Describe the characteristic of your environment:

- Commit: 05a2f3f2b813c00bc73500f45349c0306cc8c967
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: Ubuntu 22.04
- GPU: Quadro RTX 6000
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

I'll open a PR for this.

### Checklist

- [v] I have checked that there is no similar issue in the repo (**required**)
- [v] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

hhansen-bdai, 2024-07-25:
Hello @j3soon, thanks for your issue + PRs! We have a PR in the final stages which removes our dependency on `yq` in favor of the Python Standard Library's `configparser` module. This should eliminate any issues caused by snap permissions. I will close this issue when that PR is merged.

RandomOakForest, 2025-01-05:
Thank you for following up on this, @hhansen-bdai. Could we close this issue now? Thanks.

j3soon, 2025-02-28:
Confirmed that `yq` is no longer used, thanks! Closing this issue.

closed 2025-02-28 by j3soon