isaac-sim/IsaacLab #691: [Question] CPU selection

abloomyt, 2024-07-16:
### Question

I am a novice in reinforcement learning and want to use isaac sim to learn. I learned the relevant configuration information from its official website. It says that the CPU model is intel Core series or Amd Ryzen series. Compared to the Intel Core i7 (9th Generation), I only used the intel xeon gold 6133, will it still perform good?

[image]

KyleM73, 2024-07-17:
What is the clock speed of the chip? If it's [this](https://www.cpu-world.com/CPUs/Xeon/Intel-Xeon%206133.html#google_vignette) then it's a tad on the slow side, which will mainly affect startup time (e.g. how long it takes to load everything into simulation). The core consideration is the GPU/vRAM, as that is what primarily determines your performance when running the simulator/training. If your GPU is sufficiently fast then this may still work fine in my opinion. For example, on GCP we had L4 GPUs and just whatever the default CPUs were and that worked fine.

abloomyt, 2024-07-18:
> What is the clock speed of the chip? If it's [this](https://www.cpu-world.com/CPUs/Xeon/Intel-Xeon%206133.html#google_vignette) then it's a tad on the slow side, which will mainly affect startup [...]

Thank you for your reply! I get it!

closed 2024-07-18 by abloomyt