isaac-sim/IsaacLab #693: [Question] Body PhysX View

ori-gadot, 2024-07-17:
In [this PR](https://github.com/isaac-sim/IsaacLab/pull/643), the body PhysX view was removed because its functionalities are fully enabled by the root PhysX view.

I am using the set_coms method to randomize the center of mass of the robot bodies. However, when I call this method using the articulation's root_physx_view property, I encounter the following error:
`
[Error] [omni.physx.tensors.plugin] Center of Mass cannot be set with GPU pipeline.
`
Previously, when I used this method via the articulation's body_physx_view, it worked successfully.

Could someone explain why the articulation body_physx_view property was removed? Additionally, is there a way to set the center of mass properly using the root_physx_view property in a GPU pipeline?

Thanks!

AlanSunHR, 2024-07-23:
It seems that they have already fixed this issue: #641 
But the function still cannot be used after I updated Isaac Lab. I guess the issue is related to the backend of Isaac Sim, so sadly we might have to wait util the next release of Isaac Sim :(

ori-gadot, 2024-07-24:
Hi @AlanSunHR ,
Thank you for your response. I wasn't aware of the previous issue you mentioned. The function worked perfectly as long as the deprecated property body_physx_view was present in the articulation, making it a valid method. Now that I'm aware there's a pending fix to the root_physx_view set_coms method, I'll try using it again in the next release.

Mayankm96, 2024-07-28:
This should be fixed now in the latest Isaac Sim release 4.1. Please check it out @ori-gadot

ori-gadot, 2024-07-31:
@Mayankm96 I confirm this issue is fixed using IsaacSim v4.1.0

closed 2024-07-31 by ori-gadot