isaac-sim/IsaacLab #694: [Question]Imitation learning exisisting scripts

lonely2silence, 2024-07-17:
Hey there. I am trying to using isaac lab to do the imitation learning work. I see in the "exsisting script", you already has the robomimic in and give out some codes to train and implement the model, which is very helpful. But by using the giving codes and progresses(even more datas collected in provided environment), the demo do not looking good. I am wondering do you have a sucssessful result by using those models, or you just have them in? BTW, if I want to change it to my own model, and have a little change of the task, is that possible?
Have a nice day.

RandomOakForest, 2025-01-05:
Our apologies we couldn't follow up in a timely manner. We have tagged this to improve on our documentation. The team is currently working in new tutorials that will be available in the near future. Thank you for your interest in Isaac Lab.

AntoineRichard, 2026-08-27:
@alexinrobotics, thanks for asking — this nudged a real documentation gap.

The tutorials promised in the thread have shipped: there's now a full [Imitation Learning section](https://isaac-sim.github.io/IsaacLab/release/3.0.0/source/overview/imitation-learning/index.html) covering teleoperation and data collection, Isaac Lab Mimic data augmentation, humanoid imitation, and SkillGen — including published robomimic BC checkpoints and a `robust_eval` workflow, so you can see what results to expect. Swapping the task is a `--task`/`--algo` flag; other training frameworks are possible but not walked through step by step.

Closing as addressed — please open a focused issue if a specific step doesn't work for you.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard