isaac-sim/IsaacLab #701: [Bug Report] [Warning] [omni.client.python] Detected a blocking function. This will cause hitches or hangs in the UI. Please switch to the async version:

zhanglixuan0720, 2024-07-19:
### Describe the bug

I run the demo script from `source/standalone/demos/quadrupeds.py` in **VSCode integrated Terminal (command line & debugger)**, and get the warnings "[omni.client.python] Detected a blocking function. This will cause hitches or hangs in the UI. Please switch to the async version:"

The python program is blocked and does not run as expected.

This problem does not appears when I run this script in **system Terminal.** 

### Steps to reproduce

1. Open the VSCode at the IsaacLab root directory: `cd /path/to/IsaacLab & code .`
2. Activate Conda virtual environment and setup the conda_python: `conda activate isaaclab & source /home/sim/.local/share/ov/pkg/isaac-sim-4.0.0/setup_conda_env.sh`
3. run the demo python script: `python source/standalone/demos/quadrupeds.py`

```
2024-07-19 09:02:26 [47,650ms] [Warning] [carb] Client omni.stageupdate.plugin has acquired [omni::hydra::IOmniHydra v2.0] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
2024-07-19 09:02:40 [61,599ms] [Warning] [omni.client.python] Detected a blocking function. This will cause hitches or hangs in the UI. Please switch to the async version:
  File "/home/sim/anaconda3/envs/isaaclab/lib/python3.10/runpy.py", line 196, in _run_module_as_main
  File "/home/sim/anaconda3/envs/isaaclab/lib/python3.10/runpy.py", line 86, in _run_code
  File "/home/sim/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/__main__.py", line 39, in <module>
  File "/home/sim/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 430, in main
  File "/home/sim/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 284, in run_file
  File "/home/sim/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 321, in run_path
  File "/home/sim/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 135, in _run_module_code
  File "/home/sim/.vscode/extensions/ms-python.debugpy-2024.8.0-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 124, in _run_code
  File "/home/sim/nav/IsaacLab/source/standalone/demos/quadrupeds.py", line 190, in <module>
  File "/home/sim/nav/IsaacLab/source/standalone/demos/quadrupeds.py", line 181, in main
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/simulation_context.py", line 389, in reset
  File "/home/sim/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 622, in reset
  File "/home/sim/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 590, in initialize_physics
  File "/home/sim/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 887, in play
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/simulation_context.py", line 460, in render
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/asset_base.py", line 92, in <lambda>
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/asset_base.py", line 263, in _initialize_callback
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 805, in _initialize_impl
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 960, in _process_actuators_cfg
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/actuators/actuator_net.py", line 50, in __init__
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/utils/assets.py", line 125, in read_file
  File "/home/sim/.local/share/ov/pkg/isaac-sim-4.0.0/kit/extscore/omni.client/omni/client/__init__.py", line 610, in read_file
```

### System Info

Describe the characteristic of your environment:

- Commit: main
- Isaac Sim Version: 4.0.0-rc.21+4.0.13872.3e3cb0c9.gl
- OS: Ubuntu 20.04
- GPU: NVIDIA GeForce RTX 3080
- CUDA: 11.7
- GPU Driver: 550.54.14

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
Similar issue exists, but seemly is not applicable for me.
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo
I am not sure

zhanglixuan0720, 2024-07-19:
This problem can be temporally solved by changing the `console` parameter in `.vscode/launch.json` to `externalTerminal` as follows:
```
// This file is a template and is automatically generated by the setup_vscode.py script.
// Do not edit this file directly.
// 
// Generated from: /home/sim/nav/IsaacLab/.vscode/tools/launch.template.json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Python: Current File",
            "type": "python",
            "request": "launch",
            "program": "${file}",
            "console": "integratedTerminal",
            // "console": "externalTerminal",
        },
        {
            "name": "Python: Attach (windows-x86_64/linux-x86_64)",
            "type": "python",
            "request": "attach",
            "port": 3000,
            "host": "localhost"
        },
        {
            "name": "Python: Train Environment",
            "type": "python",
            "request": "launch",
            "args" : ["--task", "Isaac-Reach-Franka-v0", "--headless"],
            "program": "${workspaceFolder}/source/standalone/workflows/rsl_rl/train.py",
            "console": "integratedTerminal"
        },
        {
            "name": "Python: Play Environment",
            "type": "python",
            "request": "launch",
            "args" : ["--task", "Isaac-Reach-Franka-v0", "--num_envs", "32"],
            "program": "${workspaceFolder}/source/standalone/workflows/rsl_rl/play.py",
            "console": "integratedTerminal"
        }
    ]
}

```
However, I still expect some elegant solution for **integratedTerminal**.

RandomOakForest, 2025-01-05:
Our apologies we couldn't follow up in a timely manner. Please reopen the issue if you still see this using Isaac Sim 4.2 and Isaac Lab 1.4.0. Thank you.

closed 2025-01-05 by RandomOakForest