isaac-sim/IsaacLab #703: [Bug Report] Conda environment not being setup correctly

j3soon, 2024-07-19:
### Describe the bug

Conda environment setup script reports file not found error.

### Steps to reproduce

Following the [conda environment setup docs](https://isaac-sim.github.io/IsaacLab/source/setup/installation/binaries_installation.html#setting-up-the-conda-environment-optional):

```
$ git clone git@github.com:isaac-sim/IsaacLab.git
$ cd IsaacLab
$ ln -s path_to_isaac_sim _isaac_sim
$ ./isaaclab.sh --conda
[INFO] Using default conda environment name: isaaclab
[INFO] Creating conda environment named 'isaaclab'...
...
$ conda activate isaaclab
/home/johnson/IsaacLab/_isaac_sim/setup_conda_env.sh:.:20: no such file or directory: /home/johnson/IsaacLab/setup_python_env.sh
```

### System Info

Describe the characteristic of your environment:

- Commit: 05a2f3f2b813c00bc73500f45349c0306cc8c967
- Isaac Sim Version: 4.0.0-rc.21+4.0.13872.3e3cb0c9.gl
- OS: Ubuntu 22.04
- GPU: Quadro RTX 6000
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

I'll open a PR for this.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

j3soon, 2025-02-28:
Checking with the latest Isaac Lab v2.0.1 release (with Isaac Sim 4.5.0), the `./isaaclab.sh --conda` can run without error. However, another immediate error occurs:

```
$ git clone git@github.com:isaac-sim/IsaacLab.git
$ cd IsaacLab
$ ln -s path_to_isaac_sim _isaac_sim
$ ./isaaclab.sh --conda
[INFO] Using default conda environment name: env_isaaclab
[INFO] Creating conda environment named 'env_isaaclab'...
...
$ conda activate env_isaaclab
$ ./isaaclab.sh --install
[INFO] Installing extensions inside the Isaac Lab repository...
find: ‘/home/johnson/IsaacLab/_isaac_sim/source’: No such file or directory
```

<hr>

This is due to the `${BASH_SOURCE[0]}` used in the `./isaaclab.sh` script to determine `$ISAACLAB_PATH` has been incorrectly changed after activating the conda environment:

```
$ env | grep BASH_SOURCE
$ conda activate env_isaaclab
$ env | grep BASH_SOURCE
/home/johnson/IsaacLab/_isaac_sim/setup_python_env.sh
```

causing the `$ISAACLAB_PATH` to incorrectly point to `$HOME/IsaacLab/_isaac_sim` (instead of the correct `$HOME/IsaacLab`)

The `BASH_SOURCE` environment variable is incorrectly modified by Isaac Sim 4.5.0 upstream script `~/isaacsim/setup_conda_env.sh`. Removing the `export` line in `setup_conda_env.sh` fixes the issue:

```diff
#!/bin/bash
SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
if [ -n "$ZSH_VERSION" ]; then
    SCRIPT_DIR="$( cd "$( dirname "$0" )" && pwd )"
-   export BASH_SOURCE=$SCRIPT_DIR/setup_python_env.sh
fi
MY_DIR="$(realpath -s "$SCRIPT_DIR")"
```

On further investigation, the reason of introducing the `export` line above is a non-ideal fix for `~/isaacsim/setup_python_env.sh` to determine its filepath when using ZSH. To fix this issue, Isaac Sim v4.0.0 -> v4.1.0 introduces this `export` hack.

As Isaac Sim v4.1.0 -> v4.2.0 includes a correct patch that fixed `setup_python_env.sh`, this hack is no longer needed after v4.2.0. This patch looks something like:

```
- SCRIPT_DIR="$(dirname "${BASH_SOURCE}")"
+ SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
```

I'll keep this issue open for now, and will file a upstream issue to request removal of the `export` line in Isaac Sim.

<hr>

Potentially related issues:
- https://github.com/isaac-sim/IsaacLab/issues/1490
- https://github.com/isaac-sim/IsaacLab/issues/516

j3soon, 2025-02-28:
Aside from applying the fix for `~/isaacsim/setup_conda_env.sh`:

```diff
#!/bin/bash
SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
if [ -n "$ZSH_VERSION" ]; then
    SCRIPT_DIR="$( cd "$( dirname "$0" )" && pwd )"
-   export BASH_SOURCE=$SCRIPT_DIR/setup_python_env.sh
fi
MY_DIR="$(realpath -s "$SCRIPT_DIR")"
```

The `isaaclab.sh` script should also be patched too (according to https://github.com/isaac-sim/IsaacLab/pull/707):

```diff
        printf '%s\n' \
            '# for Isaac Sim' \
-           'source '${isaacsim_setup_conda_env_script}'' \
+           'cd '${ISAACLAB_PATH}'/_isaac_sim' \
+           'source setup_conda_env.sh' \
+           'cd $OLDPWD' \
            '' >> ${CONDA_PREFIX}/etc/conda/activate.d/setenv.sh
    fi
```

I wonder if we should re-open PR https://github.com/isaac-sim/IsaacLab/pull/707? (or submit the same PR again?)

Xunmenggod, 2025-04-22:
> Aside from applying the fix for `~/isaacsim/setup_conda_env.sh`:
> 
> #!/bin/bash
> SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
> if [ -n "$ZSH_VERSION" ]; then
>     SCRIPT_DIR="$( cd "$( dirname "$0" )" && pwd )"
> -   export BASH_SOURCE=$SCRIPT_DIR/setup_python_env.sh
> fi
> MY_DIR="$(realpath -s "$SCRIPT_DIR")"
> The `isaaclab.sh` script should also be patched too (according to [#707](https://github.com/isaac-sim/IsaacLab/pull/707)):
> 
>         printf '%s\n' \
>             '# for Isaac Sim' \
> -           'source '${isaacsim_setup_conda_env_script}'' \
> +           'cd '${ISAACLAB_PATH}'/_isaac_sim' \
> +           'source setup_conda_env.sh' \
> +           'cd $OLDPWD' \
>             '' >> ${CONDA_PREFIX}/etc/conda/activate.d/setenv.sh
>     fi
> I wonder if we should re-open PR [#707](https://github.com/isaac-sim/IsaacLab/pull/707)? (or submit the same PR again?)

Hi, I met the same problem when I tried to install the Isaac lab with isaacsim4.5 using zsh. The error says could not find the source file under _isaac_sim folder. And I tried to follow your instructions to modify the isaaclab.sh and ~/isaacsim/setup_conda_env.sh. The problem occurs again. Could you help with fixing this installing issue?

j3soon, 2025-04-22:
Hi @Xunmenggod, did you linked the isaacsim folder as mentioned here? https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html#creating-the-isaac-sim-symbolic-link

Xunmenggod, 2025-04-22:
> Hi @Xunmenggod, did you linked the isaacsim folder as mentioned here? https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html#creating-the-isaac-sim-symbolic-link

Thanks for your reply. I did the step of soft link for isaac sim folder. But the same problem that could not find the source folder occurs again.

Xunmenggod, 2025-04-27:
Hi, I found out that the Isaac lab path env is correctly fixed after re-opening a terminal to install the Isaac lab. But there is a warning to set up the VS Code settings. [WARN] Could not find Isaac Sim VSCode settings: /home/xm-laptop/Code/issac/IsaacLab/_isaac_sim/.vscode/settings.json.
	This will result in missing 'python.analysis.extraPaths' in the VSCode
	settings, which limits the functionality of the Python language server.
	However, it does not affect the functionality of the Isaac Lab project.
	We are working on a fix for this issue with the Isaac Sim team.

And I try to look for the .vscode folder in my isaacsim folder, I found that there is no .vscode folder under my isaacsim. My Isaacsim is installed by pre-built binaries.

csirois14, 2025-04-30:
I have the same problem. It think it's because the scrip's directory is not found correctly with zsh. 
I fixed it by replacing line 19 of `isaaclab.sh`
```
export ISAACLAB_PATH="$( cd "$( dirname "${BASH_SOURCE[0]}" )" &> /dev/null && pwd )"
```
to
```
if [ -n "$ZSH_VERSION" ]; then
    export ISAACLAB_PATH="$( cd "$( dirname "$0" )" &> /dev/null && pwd )"
else
    export ISAACLAB_PATH="$( cd "$( dirname "${BASH_SOURCE[0]}" )" &> /dev/null && pwd )"
fi
```

And replacing the line  
```
SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
```
in  `_isaac_sim/setup_conda_env.sh` and `_isaac_sim/setup_python_env.sh` to
```
# Determine the script directory
if [ -n "$ZSH_VERSION" ]; then
    SCRIPT_DIR="$( cd "$( dirname "$0" )" && pwd )"
else
    SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
fi
```

You then need to deactivate and reactivate your conda env. Hope this helps!

Leex1, 2025-05-15:
I resolved this issue by modifying both `setup_conda_env.sh` and `setup_python_env.sh`.
The key point is that the original logic uses `SCRIPT_DIR` based on the runtime working directory rather than the actual script location, which causes incorrect path resolution when sourced from another shell or directory.

Here’s the fix I applied:

```bash
if [ -n "$ZSH_VERSION" ]; then
    SCRIPT_PATH="${(%):-%x}"
else
    SCRIPT_PATH="${BASH_SOURCE[0]}"
fi

SCRIPT_DIR="$( cd "$( dirname "$SCRIPT_PATH" )" && pwd )"
```
This ensures that `SCRIPT_DIR` always correctly refers to the directory where the script is located, regardless of the current terminal working directory or whether it's being sourced from bash or zsh.

StafaH, 2026-09-24:
Hi, we have now migrated to a uv workflow and recommend all users to migrate as of IsaacLab 3.0. Please try out uv and let us know if there are any issues.

closed 2026-09-24 by StafaH