isaac-sim/IsaacLab #704: [Question] Can I have different number of objects in each sub-environment?

QuanyiLi, 2024-07-19:
### Question

Hi saacLab team,

Thank you for this great work! I got a question about the parallel training.

As far as I know, parallel training is launched by using the `clone` method, which replicates the same scene thousands of times with domain randomization applied to each replicated scene. As a result, the objects in the scene can have different initial conditions. 

However, it seems that is not allowed to have different objects in each sub-environment so we can not vary the number of objects (correct me if I am wrong). This prevents researchers from doing scene-level parallel training, where each scene consists of a different number/types of objects. For example, if I have a 1K room in different layouts, and I want to train navigation policy in it, I can not load each room to a single sub-environment as they have different layouts/objects in each room. So I need to fall back to using traditional multi-processing as a workaround.

I wonder if it is easy to have this feature supported. I believe it can bring much more speed-up, compared to doing the parallelization with multi-processing.

hutslib, 2024-07-25:
I meet the same problem and have the same need。 
Is there any way to workaround this?

QuanyiLi, 2024-07-25:
@hutslib A simple workaround that comes to my mind is first to find all objects you need for all sub-environments. After this, the `clone` would copy this environment with all assets. Then for each sub-environment, you can put objects that you don't need in a place that your agent can not observe, i.e. underground as if the training environment is composed of only the objects observable. This allows you to simulate parallel environments with different numbers of objects at the same time, though all sub-environments still have the same number of objects.

animesh-garg, 2024-07-25:
> @hutslib A simple workaround that comes to my mind is first to find all objects you need for all sub-environments. After this, the `clone` would copy this environment with all assets. Then for each sub-environment, you can put objects that you don't need in a place that your agent can not observe, i.e. underground as if the training environment is composed of only the objects observable. This allows you to simulate parallel environments with different numbers of objects at the same time, though all sub-environments still have the same number of objects.
> 
> (It is just a guess, I don't verify this yet

Yes I think this should be a viable solution. Basically the idea is that each env has the same number of entities preallocated which enables the tensorAPI to function correctly.

AntoineRichard, 2026-08-27:
Good question, @QuanyiLi — and thanks for proposing the workaround yourself.

The answer in the thread still holds: batched physics views need the same number of entities in every sub-environment, so pre-allocating the maximum count and parking the unused objects out of view (as @animesh-garg confirmed) remains the supported pattern. `RigidObjectCollection` is the primitive for managing many objects in that batched layout, and if you want each env to get a *different* asset (rather than a different count), `MultiAssetSpawnerCfg` / `spawn_multi_asset` now covers that.

Closing as answered — open a new issue if the multi-asset spawner doesn't cover your case.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard