isaac-sim/IsaacLab #711: Didn't find wheel for isaacsim-rl 4.0.0.0

WangYCheng23, 2024-07-20:
I want to create a conda environment for using isaaclab in a server. when I use pip install isaacsim-rl isaacsim-replicator isaacsim-extscache-physics isaacsim-extscache-kit-sdk isaacsim-extscache-kit isaacsim-app packages, there is a error shown below
```
pip install isaacsim-rl isaacsim-replicator isaacsim-extscache-physics isaacsim-extscache-kit-sdk isaacsim-extscache-kit isaacsim-app --extra-index-url https://pypi.nvidia.com
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple/, https://pypi.nvidia.com
Collecting isaacsim-rl
  Using cached https://pypi.tuna.tsinghua.edu.cn/packages/61/f2/ee3749570ad841dfaf926906eda5a88cb0200fcc2258685ae700781a86c8/isaacsim_rl-4.0.0.0.tar.gz (673 bytes)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [53 lines of output]
      INFO:nvidia-stub:Testing wheel isaacsim_rl-4.0.0.0-cp310-none-manylinux_2_34_x86_64.whl against tag cp310-none-manylinux_2_34_x86_64
      INFO:nvidia-stub:Testing wheel isaacsim_rl-4.0.0.0-cp310-none-win_amd64.whl against tag cp310-none-win_amd64
        File "/tmp/pip-build-env-gmqrne3u/overlay/lib/python3.10/site-packages/nvidia_stub/wheel.py", line 177, in download_wheel
          return download_manual(wheel_directory, distribution, version)
        File "/tmp/pip-build-env-gmqrne3u/overlay/lib/python3.10/site-packages/nvidia_stub/wheel.py", line 144, in download_manual
          raise RuntimeError(f"Didn't find wheel for {distribution} {version}")
      Traceback (most recent call last):
        File "/tmp/pip-build-env-gmqrne3u/overlay/lib/python3.10/site-packages/nvidia_stub/wheel.py", line 177, in download_wheel
          return download_manual(wheel_directory, distribution, version)
        File "/tmp/pip-build-env-gmqrne3u/overlay/lib/python3.10/site-packages/nvidia_stub/wheel.py", line 144, in download_manual
          raise RuntimeError(f"Didn't find wheel for {distribution} {version}")
      RuntimeError: Didn't find wheel for isaacsim-rl 4.0.0.0
      
      During handling of the above exception, another exception occurred:
      
      Traceback (most recent call last):
        File "/home/lixiaolin/work/anaconda3/envs/isaaclab_master/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/lixiaolin/work/anaconda3/envs/isaaclab_master/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/lixiaolin/work/anaconda3/envs/isaaclab_master/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
        File "/tmp/pip-build-env-gmqrne3u/overlay/lib/python3.10/site-packages/nvidia_stub/buildapi.py", line 29, in build_wheel
          return download_wheel(pathlib.Path(wheel_directory), config_settings)
        File "/tmp/pip-build-env-gmqrne3u/overlay/lib/python3.10/site-packages/nvidia_stub/wheel.py", line 179, in download_wheel
          report_install_failure(distribution, version, exception_context)
        File "/tmp/pip-build-env-gmqrne3u/overlay/lib/python3.10/site-packages/nvidia_stub/error.py", line 63, in report_install_failure
          raise InstallFailedError(
      nvidia_stub.error.InstallFailedError:
      *******************************************************************************
      
      The installation of isaacsim-rl for version 4.0.0.0 failed.
      
      This is a special placeholder package which downloads a real wheel package
      from https://pypi.nvidia.com. If https://pypi.nvidia.com is not reachable, we
      cannot download the real wheel file to install.
      
      You might try installing this package via
      ```
      $ pip install --extra-index-url https://pypi.nvidia.com isaacsim-rl
      ```
      
      Here is some debug information about your platform to include in any bug
      report:
      
      Python Version: CPython 3.10.14
      Operating System: Linux 4.15.0-45-generic
      CPU Architecture: x86_64
      Driver Version: 535.54
      CUDA Version: 12.2
      
      *******************************************************************************
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```
Anyone can help me solve that?

Mayankm96, 2024-07-22:
Is this Ubuntu 20.04? The pip binaries don't work on that since the default GLIBC is 2.31, but we need a minimum of 2.35.

WangYCheng23, 2024-07-22:
> Is this Ubuntu 20.04? The pip binaries don't work on that since the default GLIBC is 2.31, but we need a minimum of 2.35.

Hi~

The operating system running on my server is Ubuntu 18.04.6 LTS (Bionic Beaver).

As I do not have root privileges, I attempted to update the GLIBC library by setting the LD_LIBRARY_PATH environment variable to point to my custom GLIBC version 3.27. However, upon executing ldd --version, I encountered errors.

Given the issues with this approach, I decided to abandon it and instead transferred my local isaac-sim-4.0.0 folders  to the remote server, subsequently linking it to the isaaclab/__isaac_sim. Despite these actions, the demo script still fails to execute as expected.

I intend to utilize the remote server for training purposes due to its availability of 8 GPUs, which significantly outmatches the resources at my disposal locally. Currently, I am successfully running Isaac Sim and Isaac Lab on my local machine, which is running Ubuntu 22.04.

Mayankm96, 2024-07-23:
I don't think Isaac Sim supports Ubuntu 18.04. You'd need to use Ubuntu 20.04 or above.