isaac-sim/IsaacLab #719: [Question] camera can see robot in diffferent envs

hutslib, 2024-07-24:
I run the tutorial    
./isaaclab.sh -p source/standalone/tutorials/04_sensors/add_sensors_on_robot.py -num_evs=10

The camera mounted on the robot can see other robots. I think robots should isolate in each environment and camera should not see robot in other env.

Mayankm96, 2024-07-29:
Unfortunately, this is a current limitation of the rendering engine in Isaac Sim. The team is working on resolving this issue. However, as a quick fix, you could place walls around the environment so the cameras don't see other environments. Alternately (depending on the use-case), you can use the [RayCasterCamera](https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sensors.html#omni.isaac.lab.sensors.RayCasterCamera).

Ericonaldo, 2025-05-14:
Has this problem been resolved? This is important for visual-related projects.

Nate711, 2025-06-23:
@Mayankm96 @kellyguo11 Is there any work on fixing this issue? Thanks!

StafaH, 2026-09-25:
Thanks @hutslib for reporting that the robot-mounted camera could see robots in other environments. Isaac RTX now enables per-environment scene partitioning by default: `IsaacRtxRendererCfg` enables it, and `IsaacRtxRenderer.prepare_stage` assigns matching partition tokens to each environment and its cameras. This landed in https://github.com/isaac-sim/IsaacLab/pull/7053. The current rendering tests for articulated robots and cross-environment visibility pass. Please reopen if the tutorial still reproduces this on a current build.

closed 2026-09-25 by StafaH