isaac-sim/IsaacLab #739: [Bug Report] ModuleNotFoundError: No module named 'omni.kit.usd'

HzfFrank, 2024-07-27:
When I try to use `python source\standalone\tutorials\00_sim\create_empty.py` to verify whether the isaac lab has been installed successfully, it shows the following error:

```
2024-07-27 01:45:54 [22,506ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
        [isaaclab.python-1.0.0] dependency: (name: 'omni.isaac.menu', tag: '', version: '', exact: 0, optional: 0) can't be satisfied. Available versions:
         (none found)

2024-07-27 01:45:54 [22,508ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
  File "C:\Users\frank\IsaacLab\source\standalone\tutorials\00_sim\create_empty.py", line 29, in <module>
    app_launcher = AppLauncher(args_cli)
  File "C:\Users\frank\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py", line 113, in __init__
    self._create_app()
  File "C:\Users\frank\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py", line 532, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "C:\Users\frank\IsaacLab\_isaac_sim/exts/omni.isaac.kit\omni\isaac\kit\simulation_app.py", line 195, in __init__
    from .utils import create_new_stage, open_stage, set_livesync_stage
  File "C:\Users\frank\IsaacLab\_isaac_sim/exts/omni.isaac.kit\omni\isaac\kit\utils.py", line 17, in <module>
    from omni.kit.usd import layers
ModuleNotFoundError: No module named 'omni.kit.usd'
2024-07-27 01:45:54 [22,551ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

It seems the wrong comes out due to the lack of `omni.kit.usd` file, which should has been installed with isaac sim. And actually I cant find such a file in the `isaac sim` path. Did I lose something when installing the isaac sim?

- Isaac 2023.1.1
- OS: Win 10
- GPU: RTX 3050
- CUDA: 12.1
- GPU Driver: 531.41

HzfFrank, 2024-07-28:
It has been solved alter using isaac sim 4.0.0

closed 2024-07-28 by HzfFrank

HzfFrank, 2024-07-30:
> I am facing the same problem. Can you explain how you fixed it?

Hi, my way was install isaac sim 4.0.0 (but i didn't uninstall isaac sim 2023.1.1), then delete the original isaac lab and its conda environment should be deleted, then you can reinstall another isaac lab following the official guide (need to be linked with isaac sim 4.0.0). For me, the important thing is the version of isaac sim. Although isaac lab seems to indicate that they can still support the 2023.1.1 version of isaac sim, according to many people's feedback, this is not possible. Only isaac sim version 4.0.0 and above can support isaac lab normally. As for whether directly deleting the already installed isaac lab is the best way, I am not sure, but it is definitely feasible. The above is my way of dealing with it, I hope it can help you

Mayankm96, 2024-07-30:
Hi,

As mentioned in the release notes of Isaac Lab 1.1, we have dropped support for Isaac Sim 2023.1.1. Our standard workflow aims to support the last two releases of Isaac Sim to ensure a smoother transition for users between those versions. Maintaining support for older releases incurs a significant maintenance cost, which is currently not sustainable for our team.

I recommend trying out the latest Isaac Sim release 4.1. I hope the issue gets resolved with it.