isaac-sim/IsaacLab #750: [Question] Official pre-trained model checkpoints

j3soon, 2024-07-29:
Hi, I wonder if there's a place where we can access all pre-trained model weights (the `.pt` files) of the Isaac Lab environments?

So far, I have only found the pre-trained weights of `ANYmal-C` on Nucleus:

[image]

and some model checkpoints provided by SKRL on HuggingFace:

https://skrl.readthedocs.io/en/latest/intro/examples.html#isaac-lab-environments

Could you please let me know if there is a specific path where all the pre-trained checkpoints are stored, or if they are indeed not provided?

Thank you!

Mayankm96, 2024-07-30:
Currently, we don't provide checkpoints for the trained models. The ANYmal one is uploaded since it is used for one of the other tasks in the framework. We can look into uploading them all in the future.

Will keep this unresolved as a reminder.

j3soon, 2025-04-27:
We now have pretrained checkpoints on Nucleus, such as http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/IsaacLab/PretrainedCheckpoints/rsl_rl/Isaac-Cartpole-v0/checkpoint.pt

This can be downloaded by the `--use_pretrained_checkpoint` flag.

Ref: https://isaac-sim.github.io/IsaacLab/main/source/overview/reinforcement-learning/rl_existing_scripts.html

closed 2025-04-27 by j3soon