isaac-sim/IsaacLab #754: How to register a manager based RL environment

jih189, 2024-07-29:
I created a new task by renaming the robot of environment Isaac-Reach-UR10-v0 to 'Robot'. However, if I call 
./isaaclab.sh -p source/standalone/workflows/skrl/train.py --task Isaac-Reach-Robot-v0, it will return the error "Environment `Isaac-Reach-Robot` doesn't exist.

Even I run the ./isaaclab.sh -p source/standalone/environments/list_envs.py, the environment 'Isaac-Reach-Robot` is not list here.

Can anyone share the way to properly register the environment?

j3soon, 2024-07-30:
I suspect you didn't registered your environment.

Consider registering it in the following file:

```
./source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/reach/config/ur_10/__init__.py
```

Potential reference: https://isaac-sim.github.io/IsaacLab/source/tutorials/03_envs/register_rl_env_gym.html#manager-based-environments

jih189, 2024-07-30:
Thanks for answer. Actually, I register my environment in template code, so they are not recognized. 

I also leave my note here.
https://jih189.github.io/isaaclab_train_play

closed 2024-07-30 by jih189

XMebius, 2024-09-04:
I met the same situation and problem. I think you need to run `./isaaclab.sh --install` to configure the whole package, after that the designed name should work, no matter in template or source project. it worked for me in such way. And after configuration, I don't have to do this configuration ever.