isaac-sim/IsaacLab #768: [Question] Reinforcement learning with camera image & robot state

Hokite, 2024-08-01:
### Question

Basic questions, related to robot learning, that are not bugs or feature requests will be closed without reply, because GitHub issues are not an appropriate venue for these.

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).

Thank you so much for the provision of your learning environment. I'm currently learning the manipulator and I want to use the camera image data for learning, but I'm currently having a problem, so I'm asking you a question.

I want to put the camera image data and the current state of the robot together in the observation, is it possible?

hutslib, 2024-08-01:
Same question for the DirectRLEnv 
for rgb observation we reshape the RGB image and cat it with other observations or just let rgb image remain in it origin shape？

balaprenithreddy, 2024-08-01:
I do have the same question,

I am trying to implement RGB camera and ray-casting lidar and am not sure, how to put them together in the observation.

Mayankm96, 2024-08-01:
This depends completely on how the learning library handles these hybrid observation spaces. I know for sure RSL-RL doesn't support this out of the box. However, `skrl` and `rl-games` might be doing something about this.

There are various ways in which you can design a hybrid observation space:

- **Same observation group**: In this case, the group must NOT concatenate the individual observation terms and return them as a dictionary (MR to make this clearer: https://github.com/isaac-sim/IsaacLab/pull/772)
- **Different observation groups**: A separate observation group for images or LiDAR data to combine all proprioception-based terms in a single group. This is probably what I would do since it makes it "easier" to deal with the different possible settings

In either case, you'll have to adapt the wrappers for those frameworks according to how they expect to handle the observation spaces.

At some point, we'll migrate to using tensordicts to make it easier (but that is on our future to-do list).

hutslib, 2024-08-01:
@Mayankm96 Hi, thanks for giving us some suggestions.
I see your enhancement code and they are done with the manage based env.
For DirectRL,  what we need to do? could you give some modification examples?
And I'm more interested in the second way **Different observation groups**.

Mayankm96, 2024-08-01:
Pretty much the same. You can return a dict from the observations computation function. However, how to handle this from the learning side depends on the library.

We are working on adding more examples in this direction, but this will probably happen later this month.

closed 2024-08-03 by Mayankm96

shaoxiang, 2024-08-12:
> ### Question
> I want to put the camera image data and the current state of the robot together in the observation, is it possible?
This is easy to implement in stable baseline3. 
Just refer to the MultiInputPolicy example, which supports various Dict Observations.
You can also check out my blog post:
https://blog.csdn.net/Vulcan_S/article/details/139915681?spm=1001.2014.3001.5501

GracefulMan, 2024-08-13:
> This is easy to implement in stable baseline3. Just refer to the MultiInputPolicy example, which supports various Dict Observations. You can also check out my blog post: [...]

However, Stable Baselines3 will put the data on the CPU, which can significantly reduce performance.

Toni-SM, 2024-08-13:
**skrl** support it, but you would have to use the [skrl's Isaac Lab example scripts](https://skrl.readthedocs.io/en/latest/intro/examples.html#nvidia-isaac-lab) and not the Isaac Lab's `train.py` script and `.yaml` configurations.
 
See https://github.com/Toni-SM/skrl/discussions/185 for more details

shaoxiang, 2024-08-14:
> However, Stable Baselines3 will put the data on the CPU, which can significantly reduce performance.

Yes, stable baseline3 is really slow to train in isaac lab. 
I will explore other frameworks if I have a chance.

PR #772 (2024-08-03): Allows having hybrid dimensional terms inside an observation group
# Description

Previously, the observation manager rigidly tried concatenating all the terms along the last dimension at construction. However, when one has hybrid-dimensional terms inside the observation group, this operation would fail, leading to an error. In principle, the concatenation should only happen when users set the attribute `concatenate_terms` to True inside the group settings.

This MR lifts the above operation and adds a proper error when users try to concatenate observation terms of different shapes. They must  [... cut]
commit 4cb968ac (2024-08-03): Allows having hybrid dimensional terms inside an observation group (#772)

# Description

Previously, the observation manager rigidly tried concatenating all the
terms along the last dimension at construction. However, when one has
hybrid-dimensional terms inside the observation group, this operation
would fail, leading to an error. In principle, the concatenation should
only happen when users set the attribute `concatenate_terms` to True
inside the group settings.

This MR lifts the above operation and adds a proper error when users try
to concatenate observation terms of different shapes. Th [... cut]