isaac-sim/IsaacLab #769: [Bug Report]  --device_id can not work

kevincheng3, 2024-08-01:
I think the problem may be related to running  Isaac Sim， my classmates and I have the same problem that prevents us from using the second graphics card

### Describe the bug

When I use command `--device 1` to run the simulation using cuda:1, the simulation cannot run. The isaacsim window will pop up but the screen is black and the simulation is stuck. However, it can run normally when command `--device 0` is used.

### Steps to reproduce
For example,  running exist scripts with `./isaaclab.sh -p source/standalone/demos/hands.py`
We can successfully run the script through `./isaaclab.sh -p source/standalone/demos/hands.py --device_id 0`

[image]

But the simulation cannot be run with `./isaaclab.sh -p source/standalone/demos/hands.py --device_id 1`
[image]

And the isaacsim window is in black screen state

### System Info

- Isaac Sim Version: 4.1.0
- OS: [Ubuntu 22.04
- GPU: RTX 4090 *2 
- CUDA:  11.8
- GPU Driver: 535.183.01
[image]

GiulioRomualdi, 2024-10-05:
I don't know if this might help but a possibility is to define `CUDA_VISIBLE_DEVICES` as discussed in 
https://discuss.pytorch.org/t/how-to-change-the-default-device-of-gpu-device-ids-0/1041

glvov-bdai, 2024-10-29:
Hi,

Have you tried headless?

closed 2025-09-14 by garylvov