isaac-sim/IsaacLab #779: UnicodeDecodeError: 'cp950' codec can't decode byte 0xe2 in position 1094: illegal multibyte sequence

Daniel70-liu, 2024-08-03:
Verify Isaac Lab installation issues :
  I encountered some problems when installing Isaac Lab. When I performed the verification, an error report came out, which made my Isaac sim unable to be opened. If you have the answer, please help me. Thank you.

I use conda virtual environment and python 3.10.14 additionally I installed:

pip install isaacsim==4.1.0.0 --extra-index-url https://pypi.nvidia.com

pip install isaacsim-extscache-physics==4.1.0.0 isaacsim-extscache-kit==4.1.0.0 isaacsim-extscache-kit-sdk==4.1.0.0 --extra-index-url https://pypi.nvidia.com

2024-08-03 08:13:30 [5,121ms] [Error] [carb.scripting-python.plugin] UnicodeDecodeError: 'cp950' codec can't decode byte 0xe2 in position 1094: illegal multibyte sequence

2024-08-03 08:13:30 [5,121ms] [Error] [carb.scripting-python.plugin] UnicodeDecodeError: 'cp950' codec can't decode byte 0xe2 in position 1094: illegal multibyte sequence

At:
  C:\Users\Daniel\anaconda3\envs\isaaclab\lib\json\__init__.py(293): load
  c:/users/daniel/anaconda3/envs/isaaclab/lib/site-packages/isaacsim/exts/omni.isaac.sensor/omni/isaac/sensor/scripts/menu.py(212): __init__
  c:/users/daniel/anaconda3/envs/isaaclab/lib/site-packages/isaacsim/exts/omni.isaac.sensor/omni/isaac/sensor/scripts/extension.py(37): on_startup
  C:\Users\Daniel\anaconda3\envs\isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(164): _startup_ext
  C:\Users\Daniel\anaconda3\envs\isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(224): startup
  C:\Users\Daniel\anaconda3\envs\isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  C:\Users\Daniel\anaconda3\envs\isaaclab\Lib\site-packages\isaacsim\exts\omni.isaac.kit\omni\isaac\kit\simulation_app.py(405): _start_app
  C:\Users\Daniel\anaconda3\envs\isaaclab\Lib\site-packages\isaacsim\exts\omni.isaac.kit\omni\isaac\kit\simulation_app.py(199): __init__
  C:\Users\Daniel\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py(542): _create_app
  C:\Users\Daniel\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py(113): __init__
  C:\Users\Daniel\IsaacLab\source\standalone\tutorials\00_sim\create_empty.py(29): <module>

2024-08-03 08:13:30 [5,124ms] [Error] [omni.ext.plugin] [ext: omni.isaac.sensor-12.7.1] Failed to startup python extension.

[image]

Mayankm96, 2024-08-03:
Could you please provide more info about your machine? We are unable to reproduce this error on our end.

Daniel70-liu, 2024-08-04:
I first installed Isaac Sim in Omniverse, and then used the binary file to install Isaac Lab. When I implemented the installation verification, this error occurred.
[image]

Daniel70-liu, 2024-08-04:
My display card driver version is 555.99, the display card is RTX3090, and I use the virtual environment conda.

Everything installed in the conda environment.
# Name                    Version                   Build  Channel
absl-py                   2.1.0                    pypi_0    pypi
boto3                     1.34.153                 pypi_0    pypi
botocore                  1.34.153                 pypi_0    pypi
bzip2                     1.0.8                h2bbff1b_6
ca-certificates           2024.7.2             haa95532_0
certifi                   2024.7.4                 pypi_0    pypi
charset-normalizer        3.3.2                    pypi_0    pypi
click                     8.1.7                    pypi_0    pypi
cloudpickle               3.0.0                    pypi_0    pypi
colorama                  0.4.6                    pypi_0    pypi
contourpy                 1.2.1                    pypi_0    pypi
cycler                    0.12.1                   pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
docker-pycreds            0.4.0                    pypi_0    pypi
farama-notifications      0.0.4                    pypi_0    pypi
filelock                  3.15.4                   pypi_0    pypi
fonttools                 4.53.1                   pypi_0    pypi
fsspec                    2024.6.1                 pypi_0    pypi
gitdb                     4.0.11                   pypi_0    pypi
gitpython                 3.1.43                   pypi_0    pypi
grpcio                    1.65.2                   pypi_0    pypi
gym                       0.23.1                   pypi_0    pypi
gym-notices               0.0.8                    pypi_0    pypi
gymnasium                 0.29.0                   pypi_0    pypi
h5py                      3.11.0                   pypi_0    pypi
hidapi                    0.14.0.post2             pypi_0    pypi
idna                      3.7                      pypi_0    pypi
imageio                   2.34.2                   pypi_0    pypi
imageio-ffmpeg            0.5.1                    pypi_0    pypi
isaacsim                  4.1.0.0                  pypi_0    pypi
isaacsim-app              4.1.0.0                  pypi_0    pypi
isaacsim-asset            4.1.0.0                  pypi_0    pypi
isaacsim-benchmark        4.1.0.0                  pypi_0    pypi
isaacsim-code-editor      4.1.0.0                  pypi_0    pypi
isaacsim-core             4.1.0.0                  pypi_0    pypi
isaacsim-cortex           4.1.0.0                  pypi_0    pypi
isaacsim-example          4.1.0.0                  pypi_0    pypi
isaacsim-extscache-kit    4.1.0.0                  pypi_0    pypi
isaacsim-extscache-kit-sdk 4.1.0.0                  pypi_0    pypi
isaacsim-extscache-physics 4.1.0.0                  pypi_0    pypi
isaacsim-gui              4.1.0.0                  pypi_0    pypi
isaacsim-kernel           4.1.0.0                  pypi_0    pypi
isaacsim-replicator       4.1.0.0                  pypi_0    pypi
isaacsim-rl               4.1.0.0                  pypi_0    pypi
isaacsim-robot            4.1.0.0                  pypi_0    pypi
isaacsim-robot-motion     4.1.0.0                  pypi_0    pypi
isaacsim-robot-setup      4.1.0.0                  pypi_0    pypi
isaacsim-ros1             4.1.0.0                  pypi_0    pypi
isaacsim-ros2             4.1.0.0                  pypi_0    pypi
isaacsim-sensor           4.1.0.0                  pypi_0    pypi
isaacsim-storage          4.1.0.0                  pypi_0    pypi
isaacsim-template         4.1.0.0                  pypi_0    pypi
isaacsim-test             4.1.0.0                  pypi_0    pypi
isaacsim-utils            4.1.0.0                  pypi_0    pypi
jinja2                    3.1.4                    pypi_0    pypi
jmespath                  1.0.1                    pypi_0    pypi
kiwisolver                1.4.5                    pypi_0    pypi
libffi                    3.4.4                hd77b12b_1
markdown                  3.6                      pypi_0    pypi
markupsafe                2.1.5                    pypi_0    pypi
matplotlib                3.9.1                    pypi_0    pypi
moviepy                   1.0.3                    pypi_0    pypi
mpmath                    1.3.0                    pypi_0    pypi
networkx                  3.3                      pypi_0    pypi
numpy                     1.26.4                   pypi_0    pypi
omni-isaac-lab            0.20.5                   pypi_0    pypi
[... 37 lines cut ...]
smmap                     5.0.1                    pypi_0    pypi
sqlite                    3.45.3               h2bbff1b_0
stable-baselines3         2.3.2                    pypi_0    pypi
sympy                     1.13.1                   pypi_0    pypi
tensorboard               2.17.0                   pypi_0    pypi
tensorboard-data-server   0.7.2                    pypi_0    pypi
tensorboardx              2.6.2.2                  pypi_0    pypi
tensordict                0.3.2                    pypi_0    pypi
tk                        8.6.14               h0416ee5_0
toml                      0.10.2                   pypi_0    pypi
torch                     2.2.2                    pypi_0    pypi
torchvision               0.17.2                   pypi_0    pypi
tqdm                      4.66.4                   pypi_0    pypi
trimesh                   4.4.3                    pypi_0    pypi
[... cut ...]
urllib3                   2.2.2                    pypi_0    pypi
vc                        14.2                 h2eaa2aa_4
vs2015_runtime            14.29.30133          h43f2093_4
wandb                     0.12.21                  pypi_0    pypi
wcwidth                   0.2.13                   pypi_0    pypi
werkzeug                  3.0.3                    pypi_0    pypi
wheel                     0.43.0          py310haa95532_0
xz                        5.4.6                h8cc25b3_1
zlib                      1.2.13               h8cc25b3_1

QuanyiLi, 2024-08-10:
same error. Installed with binary with IsaacSim==4.1.0 on Windows11

gaoboyi, 2024-08-29:
> Could you please provide more info about your machine? We are unable to reproduce this error on our end.

I have the same error，how to solve it？

robotNo201, 2024-09-19:
omniverse is just a joke, still an experiment

rthaker01, 2024-09-26:
Hi @Daniel70-liu / @QuanyiLi / @gaoboyi - Are you still seeking assistance with this issue? and have you tried it with latest Isaac Sim and Isaac Lab release?

rthaker01, 2024-09-26:
Hi @robotNo201 - Can you please elaborate on your comment? what are your concerns?

StafaH, 2026-09-06:
Closing this issue. In the upcoming IsaacLab 3.0 release, all packages are pip installable and installation will be stable and effortless using uv. Feel free to give it a try early before out official release by checking out the 3.0 EA Release Branch. Enjoy!

closed 2026-09-06 by StafaH