isaac-sim/IsaacLab #782: Incorrect collision boxes after using convert_mesh.py

pbaist, 2024-08-05:
Hi,

I'm trying to use custom objects for reinforcement learning in Isaac Lab.

Following the instructions at https://isaac-sim.github.io/IsaacLab/source/how-to/import_new_asset.html#id1 I was able to run convert_mesh.py to convert my custom .obj into a .usd file usable by Isaac Lab. However, the resulting .usd exhibits incorrect collisions and nonphysical behavior (see attached video.) 

[Screencast from 2024年08月02日 16時49分02秒.webm](https://github.com/user-attachments/assets/4fa5935b-1ad4-438e-9646-84afd9b967e7)

For reference, the conversion process as done through the Isaac Sim GUI + Add -> Physics -> Rigid Body with Colliders Preset results in a .usd object that behaves correctly in physical simulation, but the resulting .usd is unusable in Isaac Lab.

Any guidance or workarounds for this issue would be appreciated.

Thanks,

Cylanx, 2024-08-05:
The same issue occurred when converting .urdf to .usd of my own robot, and i tried to add Colliders Preset also resulting in wrong prim path in Isaac Lab. I find it that the prim path saved by IsaacSim is incompatible with the correct order (which shouldn't be sub prim path under robot).

Cylanx, 2024-08-05:
> The same issue occurred when converting .urdf to .usd of my own robot, and i tried to add Colliders Preset also resulting in wrong prim path in Isaac Lab. I find it that the prim path saved by IsaacSim is incompatible with the correct order.

I've solved the order problem by save flatten as...

Mayankm96, 2024-08-06:
Hi @pbaist ,

What time of collision approximation are you setting when using the `convert_mesh.py` script? You can check the colliders of the asset in the GUI. Please check the video here for that: https://www.youtube.com/watch?v=iurSCfTyoB4

Other thing that you need to ensure is that the default prim of your created USD is on the "asset", i.e. the rigid body prim in this case. If you don't have that, then the APIs have no idea where to find the rigid body asset from.  More info on this video: https://www.youtube.com/watch?v=UyqwFWFpQfY

Cylanx, 2024-08-06:
> > > The same issue occurred when converting .urdf to .usd of my own robot, and i tried to add Colliders Preset also resulting in wrong prim path in Isaac Lab. I find it that the prim path saved by IsaacSim is incompatible with the correct order.
> > 
> > 
> > I've solved the order problem by save flatten as...
> 
> Could you go into more detail? The error I get when using an IsaacSim-converted USD in Isaac Lab is
> 
> ```
> ValueError: Not all regular expressions are matched! Please check that the regular expressions are correct: 
> 	Object: []
> Available strings: ['customusd']
> ```
> 
> I tried the "save flattened as" option but I still get the same error.
As Mayankm96 said above, you should match your prim path 'customusd' with the regular expressions used in IsaacLab. For example, the legged robot's config sets 'joint_names_expr', and my prim path should be as '/robot/FR_hip_joint' in exported usd file.
[image]

DXSheep, 2024-08-06:
> Could you go into more detail? The error I get when using an IsaacSim-converted USD in Isaac Lab is
>
> ```
> ValueError: Not all regular expressions are matched! Please check that the regular [...]

I just encountered the exact same problem. It seems that the visual shape and collision shape in the usd files generated by convert_mesh.py do not match, as shown below.

https://github.com/user-attachments/assets/2764f6df-f25f-4779-856f-7ad5b3019c93

I figured out a workaround that works for me:
1. import the obj file through Isaac Sim GUI, let's call the resulted prim "rb"
2. create a new xform called "root" and make "rb" its descendant
3. add rigid body to "root"
4. make "rb" instanceable
5. add collider to "rb"
6. save "root" to a usd file and use it in isaaclab

the key is to make the outermost prim a rigid body, as Mayankm96 said.

closed 2025-01-14 by RandomOakForest