isaac-sim/IsaacLab #783: [Question] How to get the local pose or world pose of an object in IsaacLab?

SantiDiazC, 2024-08-05:
Hi,
I was wondering how to get the pose of an object while the simulation is running (may be used as observation data: position and orientation), for example some links of an articulation (robot arm. gripper etc.) or any other spawn object inside the environment?

Thank you in advance!

cjoattesollo, 2024-08-06:
Hi, @SantiDiazC 

Did you check the code below if you make the object by RigidObjectCfg?
- ./IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/rigid_object/rigid_object_data.py

And,
- ./IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation_data.py for robot arm and gripper

Mayankm96, 2024-08-06:
Hi,

You can the rigid body transforms in different ways:

- Through the asset class, you can obtain the body poses in the world frame. For instance, [ArticulationData.body_state_w](https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.assets.html#omni.isaac.lab.assets.ArticulationData.body_state_w) and [RigidBodyData.body_state_w](https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.assets.html#omni.isaac.lab.assets.RigidObjectData.body_state_w)
- Through the [`FrameTransformer`](https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sensors.html#omni.isaac.lab.sensors.FrameTransformer) sensor, you can obtain poses between different frames. For an example, you can check the [tutorial](https://github.com/isaac-sim/IsaacLab/blob/main/source/standalone/tutorials/04_sensors/run_frame_transformer.py)

H-Hisamichi, 2024-08-06:
Hello everyone.

Can I directly obtain the local quaternion of an object in Isaac Lab? I have already retrieved the world quaternion.

Mayankm96, 2024-08-06:
Local quaternion w.r.t. what? If you mean by the environment frame, the orientation of the world frame and the environment frame coincide, so you don't need to do that.

H-Hisamichi, 2024-08-06:
Hi, @Mayankm96 

Apologies for the complex wording. 
The term 'local quaternion' refers to the orientation in the object's coordinate system (local frame). Is it correct to assume that the local environment frame and the simulated world frame are the same in Isaac Lab?

I'm trying to use the rotation information of a rolling robot (which axis and how many degrees it has rotated) in reinforcement learning, but I'm having trouble because only the simulation world frame is available in omni.isaac.lab.assets.

AshinTheAnvil, 2024-08-14:
> Hi, @Mayankm96
>
> Apologies for the complex wording. The term 'local quaternion' refers to the orientation in the object's coordinate system (local frame). Is it correct to assume that the local [...]

Hi @H-Hisamichi
 Any updates?

H-Hisamichi, 2024-08-16:
Hi, @AshinTheAnvil 

Apologies for the delayed response.
After that, it seems that the return value of body_quat_w from omni.isaac.lab.assets corresponds to observations from the local coordinate system, as in CAD, so I am using that value.

SantiDiazC, 2024-08-22:
Thank you @cjoattesollo and @Mayankm96 for your replies! 
It helped me a lot to do what I was looking for!

mikelasa, 2024-10-09:
Related to this issue, I have checked the rigid_object_data.py and I all the functions to get the object position are wrt to the world frame. I wonder if I could get the position of my object wrt to his own frame in the last iteration.

Thank you

KnuniK, 2025-04-03:
Weird question:

In my simulation, `body_state_w/body_link_state_w and body_com_state_m` stay constant, while attributes like `joint_pos` update every step.
(While watching the simuation you can clearly see that the bodies of my robot are moving)

Does somebody have an idea for a direction he/she can point me for solving this?

Shuaikang-Wang, 2025-04-15:
> Weird question:
>
> In my simulation, `body_state_w/body_link_state_w and body_com_state_m` stay constant, while attributes like `joint_pos` update every step. (While watching the simuation you can [...]

Hi, did you solve this problem?

KnuniK, 2025-04-15:
> Hi, did you solve this problem?

Yes, I did!

My Script is based on the[ Direct Workflow RL Tutorial](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/create_direct_rl_env.html).
 I added a variable in the `__int__` method of the `CartpoleEnv ` at line 69:
`self.body_state = self.cartpole.data.body_state_w ` 

However, if you access `self.body_state` in functions like  `_get_observations`/`_get_dones` or `_get_rewards`, it always returns the initial state and is never updated.

To get the current values, you need to reassign the variable inside each of those functions:
`self.body_state = self.cartpole.data.body_state_w ` 
This ensures you always work with the updated state.

I'm not sure if this is intended behavior or a bug (tested on Isaac Sim version 4.5.0).

Shuaikang-Wang, 2025-04-15:
> Yes, I did!
>
> My Script is based on the[ Direct Workflow RL Tutorial](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/create_direct_rl_env.html). I added a variable in the [...]

That works! Thank you

closed 2026-09-15 by maxkra15