isaac-sim/IsaacLab #788: [Bug Report] import omni.replicator.core as rep ModuleNotFoundError: No module named 'omni.replicator'

zhanglixuan0720, 2024-08-05:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Error in running app in --headless and --enable_cameras mode
```
Traceback (most recent call last):
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 372, in _initialize_impl
    import omni.replicator.core as rep
ModuleNotFoundError: No module named 'omni.replicator'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 64, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 271, in _initialize_callback
    self._initialize_impl()
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 374, in _initialize_impl
    raise RuntimeError(
RuntimeError: Replicator was not found for rendering. Please use --enable_cameras to enable rendering.
Traceback (most recent call last):
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 372, in _initialize_impl
    import omni.replicator.core as rep
ModuleNotFoundError: No module named 'omni.replicator'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 64, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 271, in _initialize_callback
    self._initialize_impl()
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 374, in _initialize_impl
    raise RuntimeError(
RuntimeError: Replicator was not found for rendering. Please use --enable_cameras to enable rendering.

Traceback (most recent call last):
  File "/home/sim/nav/tutorials/t-2-teleop_se2_agent.py", line 153, in <module>
    main()
  File "/home/sim/nav/tutorials/t-2-teleop_se2_agent.py", line 85, in main
    env = ManagerBasedRLEnv(env_cfg)
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 75, in __init__
    super().__init__(cfg=cfg)
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 132, in __init__
    self.load_managers()
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 112, in load_managers
    super().load_managers()
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 220, in load_managers
    self.observation_manager = ObservationManager(self.cfg.observations, self)
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/observation_manager.py", line 41, in __init__
    super().__init__(cfg, env)
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/manager_base.py", line 130, in __init__
    self._prepare_terms()
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/observation_manager.py", line 254, in _prepare_terms
    obs_dims = tuple(term_cfg.func(self._env, **term_cfg.params).shape[1:])
  File "/home/sim/nav/tutorials/mdp/observations.py", line 51, in isaac_camera_data
    sensor: CameraData = env.scene.sensors[sensor_cfg.name].data
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 170, in data
    self._update_outdated_buffers()
  File "/home/sim/nav/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 284, in _update_outdated_buffers
    outdated_env_ids = self._is_outdated.nonzero().squeeze(-1)
AttributeError: 'Camera' object has no attribute '_is_outdated'

```

### Steps to reproduce

```
python t-2-teleop_se2_agent.py --headless --enable_cameras
```
Code:
```python
# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

"""Script to run a keyboard teleoperation with Isaac Lab manipulation environments."""

"""Launch Isaac Sim Simulator first."""

import argparse

from omni.isaac.lab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="Keyboard teleoperation for Isaac Lab environments.")
parser.add_argument("--cpu", action="store_true", default=False, help="Use CPU pipeline.")
parser.add_argument(
    "--disable_fabric", action="store_true", default=False, help="Disable fabric and use USD I/O operations."
)
parser.add_argument("--num_envs", type=int, default=1, help="Number of environments to simulate.")
parser.add_argument("--device", type=str, default="keyboard", help="Device for interacting with environment")
parser.add_argument("--task", type=str, default=None, help="Name of the task.")
parser.add_argument("--sensitivity", type=float, default=1.0, help="Sensitivity factor.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(headless=args_cli.headless)
simulation_app = app_launcher.app

"""Rest everything follows."""

import gymnasium as gym
import torch

import carb

from omni.isaac.lab.devices import Se3Gamepad, Se3Keyboard, Se3SpaceMouse, Se2Keyboard

import omni.isaac.lab_tasks  # noqa: F401
from omni.isaac.lab_tasks.utils import parse_env_cfg

import omni.isaac.core.utils.prims as prim_utils
import torch
from omni.isaac.core.objects import VisualCuboid
# from omni.isaac.lab.envs import RLTaskEnv
from omni.isaac.lab.envs import ManagerBasedRLEnv
from config_vp import ViPlannerWarehouseVelCfg 
import time
import numpy as np

def main():
    """Running keyboard teleoperation with Isaac Lab manipulation environment."""
    # parse configuration
    env_cfg = ViPlannerWarehouseVelCfg()
    env = ManagerBasedRLEnv(env_cfg)
    

    # create controller
    teleop_interface = Se2Keyboard()

    # add teleoperation key for env reset
    teleop_interface.add_callback("L", env.reset)
    # print helper for keyboard
    print(teleop_interface)

    # reset environment
    env.reset()
    teleop_interface.reset()

    running_times = []
    count = 0
    # simulate environment
    print('setup ok')
    while simulation_app.is_running():
        t0 = time.perf_counter()
        # run everything in inference mode
        with torch.inference_mode():
            # get keyboard command
         
            vel_cmd = teleop_interface.advance()
            # pre-process actions
            vel_cmd = vel_cmd.reshape(1,3)
            vel_cmd = torch.from_numpy(vel_cmd)
            obs = env.step(action=vel_cmd)[0]
        t1 = time.perf_counter()
        running_times.append(t1-t0)
        count += 1 
        if count % 10 == 0:
            mean_processed_time = np.mean(running_times)
            print(f'mean: {mean_processed_time} \n frequency: {1/mean_processed_time}' )
    # close the simulator
    env.close()

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()

```

-->

### System Info

Describe the characteristic of your environment:

- Commit: main
- Isaac Sim Version: 2024
- OS: Ubuntu 20.04
- GPU: [e.g. RTX 2060 Super]
- CUDA: NVIDIA GeForce RTX 3080
- GPU Driver: 550.54.14

### Additional context
None

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
Similar but not suitable
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo
Not sure

Dhoeller19, 2024-08-05:
Hi @zhanglixuan0720,
your problem is that you don't feed the cli args back to the Applauncher, i.e. you have to do:
```app_launcher = AppLauncher(args_cli)```

zhanglixuan0720, 2024-08-05:
@Dhoeller19 Thank you!

closed 2024-08-05 by zhanglixuan0720