isaac-sim/IsaacLab #790: [Question] train from checkpoint

sizsJEon, 2024-08-06:
### Question
 
I'm trying to start with Isaac Lab, 

but after I run first training with rl_games,

I can't find a option for checkpoint of finetune method.

Is issac lab's rl_games are currently not supporting it?

Dhoeller19, 2024-08-06:
Hi @sizsJEon, 
This is supported, the instruction are at the bottom of this page: https://isaac-sim.github.io/IsaacLab/source/setup/sample.html.

sizsJEon, 2024-08-07:
The checkpoint in the example at the bottom of the page is an option when playing with **play.py**, 
and it seems to be an option that is not applied when doing train with train.py.

https://isaac-sim.github.io/IsaacLab/source/setup/sample.html
```
# install python module (for rl-games)
./isaaclab.sh -i rl_games
# run script for training
./isaaclab.sh -p source/standalone/workflows/rl_games/train.py --task Isaac-Ant-v0 --headless
# run script for playing with 32 environments
./isaaclab.sh -p source/standalone/workflows/rl_games/play.py --task Isaac-Ant-v0 --num_envs 32 --checkpoint /PATH/TO/model.pth
# run script for recording video of a trained agent (requires installing `ffmpeg`)
./isaaclab.sh -p source/standalone/workflows/rl_games/play.py --task Isaac-Ant-v0 --headless --video --video_length 200
```

arguments for train and play 
https://github.com/isaac-sim/IsaacLab/blob/d178560c7a5224e117c201edd6b34afdd5551a06/source/standalone/workflows/rl_games/train.py#L15-L28
https://github.com/isaac-sim/IsaacLab/blob/d178560c7a5224e117c201edd6b34afdd5551a06/source/standalone/workflows/rl_games/play.py#L15-L28

also load_checkpoint code is only in play.py
https://github.com/isaac-sim/IsaacLab/blob/4def7a692ba7e0b5b59e9d37266e5701e6a2f680/source/standalone/workflows/rl_games/play.py#L100-L101

you could find difference between 
https://github.com/isaac-sim/IsaacLab/blob/4def7a692ba7e0b5b59e9d37266e5701e6a2f680/source/standalone/workflows/rl_games/train.py#L145
https://github.com/isaac-sim/IsaacGymEnvs/blob/aeed298638a1f7b5421b38f5f3cc2d1079b6d9c3/isaacgymenvs/train.py#L210-L215

I have submitted a request regarding this matter. Could you please review it?

sizsJEon, 2024-08-07:
> Hi @sizsJEon, This is supported, the instruction are at the bottom of this page: https://isaac-sim.github.io/IsaacLab/source/setup/sample.html.

Hello @Dhoeller19 , 
I think the checkpoint in the example at the bottom of the page is an option when playing with **play.py**, 
and it seems to be not appliable when doing train with train.py.

https://isaac-sim.github.io/IsaacLab/source/setup/sample.html
```
# install python module (for rl-games)
./isaaclab.sh -i rl_games
# run script for training
./isaaclab.sh -p source/standalone/workflows/rl_games/train.py --task Isaac-Ant-v0 --headless
# run script for playing with 32 environments
./isaaclab.sh -p source/standalone/workflows/rl_games/play.py --task Isaac-Ant-v0 --num_envs 32 --checkpoint /PATH/TO/model.pth
# run script for recording video of a trained agent (requires installing `ffmpeg`)
./isaaclab.sh -p source/standalone/workflows/rl_games/play.py --task Isaac-Ant-v0 --headless --video --video_length 200
```

arguments for train and play 
https://github.com/isaac-sim/IsaacLab/blob/d178560c7a5224e117c201edd6b34afdd5551a06/source/standalone/workflows/rl_games/train.py#L15-L28
https://github.com/isaac-sim/IsaacLab/blob/d178560c7a5224e117c201edd6b34afdd5551a06/source/standalone/workflows/rl_games/play.py#L15-L28

also load_checkpoint code is only in play.py
https://github.com/isaac-sim/IsaacLab/blob/4def7a692ba7e0b5b59e9d37266e5701e6a2f680/source/standalone/workflows/rl_games/play.py#L100-L101

you could find difference between 
https://github.com/isaac-sim/IsaacLab/blob/4def7a692ba7e0b5b59e9d37266e5701e6a2f680/source/standalone/workflows/rl_games/train.py#L145
https://github.com/isaac-sim/IsaacGymEnvs/blob/aeed298638a1f7b5421b38f5f3cc2d1079b6d9c3/isaacgymenvs/train.py#L210-L215

I have submitted a request regarding this matter. Could you please review it?

Dhoeller19, 2024-08-07:
Ah, yes for trainin indeed :+1:

closed 2024-08-07 by sizsJEon