isaac-sim/IsaacLab #796: [Bug Report] Bug title

WangYCheng23, 2024-08-07:
### Describe the bug

To leverage my server's multi-GPU capabilities, I created a Docker container using Ubuntu 22.04 to meet the minimum GLIBC version requirements. However, upon running the isaac-lab script within this container, numerous errors surfaced. Assistance in troubleshooting and resolving these issues would be greatly appreciated.

### Steps to reproduce

1.create a docker container with image "m.daocloud.io/nvcr.io/nvidia/cuda:12.2.0-base-ubuntu22.04 "
2.build env with pip and run ./isaaclab.sh -i. everything works fine.
3.run ./isaaclab.sh -p source/standalone/tutorials/00_sim/create_empty.py --headless and then error is shown as follow

```
(isaaclab) root@be1ce1dfa6bf:/data/RL/IsaacLab# ./isaaclab.sh -p source/standalone/tutorials/00_sim/create_empty.py --headless
[INFO] Using python from: /root/anaconda3/envs/isaaclab/bin/python                                                                                                              
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: /data/RL/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.headless.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/data/RL/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/../../../../../../apps/isaaclab.python.headless.kit', '--/app/tokens/exe-path=/root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/exts', '--ext-folder', '/root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/apps', '--/renderer/activeGpu=0', '--/physics/cudaDevice=0', '--portable', '--no-window', '--/app/window/hideUi=True', '--allow-root']
Passing the following args to the base kit application:  ['--headless']
Loading user config located at: '/root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.1/user.config.json'
[Info] [carb] Logging to file: /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.1/kit_20240807_042544.log
2024-08-07 04:25:44 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.043s] [ext: omni.kit.async_engine-0.0.0] startup
[0.124s] [ext: omni.stats-1.0.1] startup
[0.125s] [ext: omni.client-1.1.0] startup
[0.162s] [ext: omni.assets.plugins-0.0.0] startup
[0.164s] [ext: omni.gpu_foundation-0.0.0] startup
[0.177s] [ext: omni.kit.pipapi-0.0.0] startup
[0.177s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.178s] [ext: omni.mtlx-0.1.0] startup
[0.178s] [ext: omni.usd.config-1.0.4] startup
[0.182s] [ext: omni.gpucompute.plugins-0.0.0] startup
[0.183s] [ext: omni.usd.libs-1.0.1] startup
2024-08-07 04:25:44 [195ms] [Error] [omni.ext.plugin] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/libgarch.so. Error: libSM.so.6: cannot open shared object file: No such file or directory (note that this may be caused by a dependent library)
2024-08-07 04:25:44 [196ms] [Error] [omni.ext.plugin] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/libglf.so. Error: libSM.so.6: cannot open shared object file: No such file or directory (note that this may be caused by a dependent library)
2024-08-07 04:25:44 [207ms] [Error] [omni.ext.plugin] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/libhdSt.so. Error: libSM.so.6: cannot open shared object file: No such file or directory (note that this may be caused by a dependent library)
2024-08-07 04:25:44 [208ms] [Error] [omni.ext.plugin] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/libhdx.so. Error: libSM.so.6: cannot open shared object file: No such file or directory (note that this may be caused by a dependent library)
2024-08-07 04:25:44 [209ms] [Error] [omni.ext.plugin] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/libhgiGL.so. Error: libSM.so.6: cannot open shared object file: No such file or directory (note that this may be caused by a dependent library)
2024-08-07 04:25:44 [209ms] [Error] [omni.ext.plugin] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/libhgiInterop.so. Error: libSM.so.6: cannot open shared object file: No such file or directory (note that this may be caused by a dependent library)
2024-08-07 04:25:44 [227ms] [Error] [omni.ext.plugin] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/libusdImagingGL.so. Error: libSM.so.6: cannot open shared object file: No such file or directory (note that this may be caused by a dependent library)
2024-08-07 04:25:44 [251ms] [Error] [omni.ext.plugin] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/libMaterialXRenderHw.so. Error: libSM.so.6: cannot open shared object file: No such file or directory (note that this may be caused by a dependent library)
2024-08-07 04:25:44 [251ms] [Error] [omni.ext.plugin] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/libMaterialXRenderGlsl.so. Error: libSM.so.6: cannot open shared object file: No such file or directory (note that this may be caused by a dependent library)
2024-08-07 04:25:44 [252ms] [Error] [carb] [Plugin: libomni.fabric.plugin.so] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/fabric/libomni.fabric.plugin.so. Error: libhdx.so: cannot open shared object file: No such file or directory (Additional information may be available by running the process with the LD_DEBUG environment variable set)
2024-08-07 04:25:44 [252ms] [Warning] [carb] Potential plugin preload failed: /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/fabric/libomni.fabric.plugin.so
2024-08-07 04:25:44 [252ms] [Error] [omni.ext.plugin] [ext: omni.usd.libs-1.0.1] failed to load native plugin: /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/fabric/libomni.fabric.plugin.so
2024-08-07 04:25:44 [253ms] [Error] [carb] [Plugin: libomni.usd-abi.plugin.so] Could not load the dynamic library from /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/deps/libomni.usd-abi.plugin.so. Error: libhdx.so: cannot open shared object file: No such file or directory (Additional information may be available by running the process with the LD_DEBUG environment variable set)
2024-08-07 04:25:44 [253ms] [Warning] [carb] Potential plugin preload failed: /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.usd.libs/bin/deps/libomni.usd-abi.plugin.so
[... cut ...]
2024-08-07 04:25:45 [992ms] [Warning] [carb] Client omni.kit.telemetry.plugin Failed to acquire interface [gpu::foundation::ISystemInfo v0.5] while unloading all plugins
2024-08-07 04:25:45 [993ms] [Warning] [omni.core.ITypeFactory] Module /root/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

- Commit: v1.1.0
- Isaac Sim Version: 4.0.1(pip)
- OS: Ubuntu 22.04
- GPU: V100*8
- CUDA: 12.2
- GPU Driver: 535.54.03

WangYCheng23, 2024-08-08:
I addressed this issue by following the guidelines provided on the Isaac Lab's Docker deployment documentation page [1]. The process involved setting up an account, creating a Docker image for Isaac Sim. Then I subsequently developing within the Isaac Lab environment encapsulated in the Docker container.

closed 2024-08-08 by WangYCheng23

hyonhojoh, 2025-05-08:
sudo apt update
sudo apt install -y libsm6 libglu1-mesa