isaac-sim/IsaacLab #798: [Proposal] Update Se3Gamepad to support new gamepad controller

kevincheng3, 2024-08-07:
### Proposal

Update Se3Gamepad to support new gamepad controller

### Motivation
I want to use `Se3Gamepad` control a quadruped robot with Xbox. It works fine when I use the old Xbox, but not the new one.

Specifically， when using the `gamepad` to execute the `teleop_se3_agent.py` script, I found that the **old** Xbox controller can work normally, but the **new** Xbox controller cannot be used. Is there a problem with my hardware, or does this library need to be updated?

### Alternatives

Are there any other libraries to support the new Xbox？

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

- [ ] Use the new Xbox to implement `teleop_se3_agent.py`

Dhoeller19, 2024-08-07:
Hi @kevincheng3,
Thanks for the report. I was able to reproduce this issue as well. This seems to be an issue with Kit since I also have the issue with the Kaya Gamebad example in Isaac Sim (https://docs.omniverse.nvidia.com/isaacsim/latest/features/external_communication/ext_omni_isaac_gamepad.html#kaya-gamepad-example). I have pinged the corresponding teams.

zxf8665905, 2024-08-18:
@kevincheng3  Which grip are you using that works? Is it wired or bluetooth?

AntoineRichard, 2026-08-27:
Thanks for the detailed report and the follow-up testing, @kevincheng3 — much appreciated.

Gamepad devices are enumerated by the underlying input layer rather than by Isaac Lab; `source/isaaclab/isaaclab/devices/gamepad/se3_gamepad.py` consumes the carb input events it receives, so when a controller isn't picked up there's no mapping we can add on this side. This was raised with the input/simulation team at the time.

Quite a bit has changed across releases since, so we're closing this one out — if a controller still isn't detected on a current Isaac Sim, please reopen with the model and connection type (USB vs Bluetooth) and we'll take another look.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard