isaac-sim/IsaacLab #800: [Question] Question about changing joint's drive type

sizsJEon, 2024-08-07:
### Question
Hi, I'm immigrating from IssacGymEnvs(Gym) and have a question about joint's drive mode,

In Gym, I was able to set the joints drive mode myself.
for example, I could make a torque-commanded robot arm attacted to velocity-commanded 4 wheel car
like 
```
dof_props['driveMode'][1:6] = gymapi.DOF_MODE_POS
dof_props['driveMode'][7:10] = gymapi.DOF_MODE_VEL
```
and If I could set pos/vel cmd by 
```
self.gym.set_dof_position_target_tensor(self.sim, gymtorch.unwrap_tensor(targets))
self.gym.set_dof_velocity_target_tensor(self.sim, gymtorch.unwrap_tensor(targets))
```

but in Isaac Lab(Lab), 
I have to make a usd from urdf and it seems to be joint drive mode is determined at there.

I looked the cartpole, unitree, universal_robots in lab_assets 
but could not find the difference between torque (cartpole) and position (unitree, universal_robots)

the point is, 
**How could I change the drive mode of the joint manually(in code)?**

https://isaac-sim.github.io/IsaacLab/source/how-to/import_new_asset.html#using-urdf-importer
it says I could change the drive configuration, but I can't find any information about it.

jsmith-bdai, 2024-08-09:
Hello! The way you can inform IsaacLab of your desired articulation type (e.g. position, velocity, force or some combination thereof)  is by setting up an associated `JointActionTerm` in your environment config's `ActionCfg`. Before that configuration step, actuators can support the various modes of control.

Here's an example with Cartpole, where we're using effort-based control: https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/classic/cartpole/cartpole_env_cfg.py#L75

Here's an example with a locomotion environment where we're using position-based control: https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/locomotion/velocity/velocity_env_cfg.py#L112

And lastly, here's an example with a manipulation environment where we're also using position-based control: https://github.com/isaac-sim/IsaacLab/blob/6451d235785780167f80c14bb9d635d20aff9b29/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/lift/lift_env_cfg.py#L91-L92

Not you can also modify the paramaters you wish to observe via the `ObservationCfg`.

Hope this helps!

sizsJEon, 2024-08-09:
> Hello! The way you can inform IsaacLab of your desired articulation type (e.g. position, velocity, force or some combination thereof) is by setting up an associated `JointActionTerm` in your [...]

since I use direct rl configuration, I have to do something else, 
but It's good to know that there are some way. 

thanks for the information and wait for other advices!

jsmith-bdai, 2024-08-09:
CC: @kellyguo11 might have more info here

2361098148, 2024-08-27:
The all actutor they provide seems to be essentially torque control, and the position control provided is also calculated through kp and kd to obtain torque for delivery. It's not like the dof_mode_pos in Isaac Gym. Do you have any solutions?

sizsJEon, 2024-08-27:
for now, I'd back to gym env unfortunately. If anything is progress, I'll share here.

2361098148, 2024-08-27:
@Mayank Mittal

RandomOakForest, 2025-01-06:
The team has been adding to this, and we have an [updated doc on actuators](https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/actuators.html). Let us know if we can be of further help.

closed 2025-01-06 by RandomOakForest