isaac-sim/IsaacLab #801: [Bug Report] 'RigidBodyView' object has no attribute 'get_accelerations'

mickeyouyou, 2024-08-07:
### Describe the bug

'RigidBodyView' object has no attribute 'get_accelerations'

### Steps to reproduce

`./isaaclab.sh -p source/standalone/tutorials/04_sensors/run_ray_caster.py`

```log
[INFO]: Setup complete...
Traceback (most recent call last):
  File "/home/lenovo/codes/IsaacLab/source/standalone/tutorials/04_sensors/run_ray_caster.py", line 148, in <module>
    main()
  File "/home/lenovo/codes/IsaacLab/source/standalone/tutorials/04_sensors/run_ray_caster.py", line 143, in main
    run_simulator(sim=sim, scene_entities=scene_entities)
  File "/home/lenovo/codes/IsaacLab/source/standalone/tutorials/04_sensors/run_ray_caster.py", line 112, in run_simulator
    balls.write_root_state_to_sim(ball_default_state)
  File "/home/lenovo/codes/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/rigid_object/rigid_object.py", line 158, in write_root_state_to_sim
    self.write_root_velocity_to_sim(root_state[:, 7:], env_ids=env_ids)
  File "/home/lenovo/codes/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/rigid_object/rigid_object.py", line 198, in write_root_velocity_to_sim
    self._data.body_acc_w[env_ids] = 0.0
  File "/home/lenovo/codes/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/rigid_object/rigid_object_data.py", line 137, in body_acc_w
    self._body_acc_w.data = self._root_physx_view.get_accelerations().unsqueeze(1)
AttributeError: 'RigidBodyView' object has no attribute 'get_accelerations'
```

### System Info

Describe the characteristic of your environment:

- Commit: [e151bb93b8d818bd359d86ba5e00f5f54f858783]
- Isaac Sim Version: 2023.1.1-rc.8+2023.1.688.573e0291.tc
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. RTX 3080]
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

Mayankm96, 2024-08-07:
Please use Isaac Sim 4.0 and above with the latest code.

We dropped the support for Isaac Sim 2023.1.1 with the IsaacLab release 1.1.

mickeyouyou, 2024-08-07:
How to get Isaac Sim 4.0, We just download  omniverse launcher and downloaded latest version is  Isaac Sim 2023.1.1.

closed 2024-08-07 by mickeyouyou

Bruce-Si, 2024-08-18:
> Please use Isaac Sim 4.0 and above with the latest code.
> 
> We dropped the support for Isaac Sim 2023.1.1 with the IsaacLab release 1.1.

I used Isaac Sim 4.0.0 but also got the error: 'RigidBodyView' object has no attribute 'get_accelerations'.
[image]

Bruce-Si, 2024-08-19:
> Good catch. We recently realized that this attribute was only added in Isaac Sim 4.1.0 :/
> 
> We will have to send a patch fix for it soon to have backwards compatibility. You can find the change made here: #760

Thank you, I switched to Isaac Sim 4.1.0 and it worked!